SmingHub/Sming #820: Sming and SDK compatability

icodk, 2016-11-29:
I just tried a fresh manual (can't use Chocolate on WindowsXP) Sming installation by downloading 
the SDK from http://programs74.ru/ which is now in version 2.2.1 and is based on the ESP8266_NONOS_SDK_V2.0.0_16_07_19
From #817 I leraned that Sming is not compatible with ESP8266_NONOS_SDK_V2.0.0_16_07_19 . However this is the version that is going to be downloaded from http://programs74.ru/
May be it should be mentioned somewhere  which version of the unofficial SDK Sming is compatible with.
Might help somebody:
To get SDK 1.5.4 download version 2.1.0 from https://github.com/CHERTS/esp8266-devkit/tags
and rename the C:\Espressif\ESP8266_SDK to C:\Espressif\ESP8266_SDK_200 and rename the C:\Espressif\ESP8266_SDK_ to  C:\Espressif\ESP8266_SDK_154

STwilight, 2016-12-28:
Renaming folders is good idea, thanks, @icodk!
@anakod, if you can, provide us with the info, please, when we can wait for the **release of Sming**, that will **support 2.0.0 SDK** version?
Thanks.

slaff, 2017-02-09:
SDK 2.0 is supported in the latest `develop` version.

closed 2017-02-09 by slaff