SmingHub/Sming #828: IOT deepSleep example for Sming

kwis2, 2016-12-07:
I am wondering how to ask ESP to lower used power. Especially how to use deep sleep to get lowest possible power. 
I wasn't able to find any example in Sming, there is complicated example of deep sleep - IOT_demo in Espressif. 
Would it be possible to make in Sming simple deepSleep demo for beginners? for instance like wake up. connect to Wifi, HTTP post data and then deepSleep.
With old issues I have found discussion about [Rapid Loader](https://github.com/pvvx/Rapid_Loader). Would it be useful to decrease total power with Sming?

lxkarthi, 2017-01-31:
deepSleep is timed. 
Call the following function in SMING project code
System.deepSleep(seconds*1000);

kwis2, 2017-01-31:
@lxkarthi thank you for your answer. It is very easy to make ESP sleep. Question is how to wake chip up. what is the fastest, or maybe lowest power! method to start communication. ie. send http post.

lxkarthi, 2017-02-04:
The wake up is timed. 
I found that you have to connect GPIO16 to RST pin to make DeepSleep work. There are many modes for deepSleep.  check deepSleep options http://sminghub.github.io/sming-api-develop/group__system.html#ga15eb21125083b655bb83f7dbd47f0987 
I haven't tried these options. 
eDSO_RF_CAL_BY_INIT_DATA = 0, ///<  RF_CAL  or  not  after  deep-sleep  wake  up, depends on init data byte 108.
eDSO_RF_CAL_ALWAYS = 1, ///< RF_CAL after deep-sleep wake up, there will be large current.
eDSO_RF_CAL_NEVER = 2, ///<  no RF_CAL after deep-sleep wake up, there will only be small current.
eDSO_DISABLE_RF = 4, ///< disable RF after deep-sleep wake up, just like modem sleep, there will be the smallest current.
 Do you want to wake up the chip by interrupts?

slaff, 2017-02-09:
Feel free to reopen the issue if it is Sming related.

closed 2017-02-09 by slaff