SmingHub/Sming #829: We need logo for the project.

slaff, 2016-12-09:
We are looking for a logo for the project. If you have suggestions make sure to put them as comments to this issue.

chall3ng3r, 2017-01-17:
My younger brother is professional logo designer. I'll ask him for few designs to select from. Will post here.

DominicMe, 2017-02-05:
A while back I made a logo for [ESP8266 Sming Framework](http://dominicm.com/esp8266-sming-framework-arch-linux/) article as I couldn't find one. The low resolution logo used at a time and now seems to be taken from a railway company in Netherlands which is a bit bizarre :)

@anakod noticed the logo and suggested I post it here. I can send vector / psd original if you decide you want to use it.

slaff, 2017-02-05:
@DominicMe Please, send the PSD as long as it does not infringe any copyright laws.

anakod, 2017-02-06:
@DominicMe glad to see you on github. I'm personally like logo designed by you, and think it can be used as official Sming Framework logo.

DominicMe, 2017-02-10:
Here is the AI vector file and high res png.

http://owncloud.dominicm.com/index.php/s/lUJ5ckZtyz1aPtt

[image]

closed 2017-02-21 by slaff

anakod, 2017-02-21:
Big thanks @DominicMe !