SmingHub/Sming #830: Increased TCP connection memory usage even with SSL disabled

gbl08ma, 2016-12-11:
Recently, SSL support was added to Sming. This is a good addition, and because it noticeably increases memory usage, it can be disabled at compilation time. However, I noticed that my programs would use more memory when linked against the develop branch, **even with SSL disabled**, so I decided to investigate. Turns out that some classes like TcpConnection just became much larger (in terms of memory usage) due to SSL, even when SSL support is disabled.

This is what the TcpConnection class holds:

```
protected:
	tcp_pcb *tcp = NULL;
	uint16_t sleep;
	uint16_t timeOut;
	bool canSend;
	bool autoSelfDestruct;
#ifdef ENABLE_SSL
	SSL *ssl = nullptr;
	SSLCTX *sslContext = nullptr;
	SSL_EXTENSIONS *ssl_ext=NULL;
#endif
	boolean useSsl = false;
	uint8_t *sslFingerprint=null;
	boolean sslConnected = false;
	uint32_t sslOptions=0;
	SSLKeyCertPair clientKeyCert;
	boolean freeClientKeyCert = false;
```

As you can see, there are lots of SSL-related stuff outside of the `ENABLE_SSL` preprocessor conditional. I tried to move these into the conditional, making the appropriate changes to TcpConnection.cpp, and quickly found out that lots of other classes (e.g. the HTTP and MQTT clients) expect some of the SSL stuff to exist even with SSL support disabled. A more thorough analysis and refactoring of the code would be needed to make non-SSL builds use as little memory as before SSL was added to Sming.

As things are now, each TCP connection consumes at least 18 more bytes of RAM than previously, with things that are useless because SSL is not enabled.

I have another idea: would it be possible to refactor SSL-enabled TCP connections into another class that inherits from TcpConnection? This would allow for non-SSL connections to use less RAM even on builds with SSL enabled. However, some more methods may need to become virtual and I'm not sure what impact this will have on performance.

slaff, 2016-12-12:
> However, I noticed that my programs would use more memory when linked against the develop branch, even with SSL disabled

Can you be more specific: which memory do you mean? Is it the flash memory, is the heap is it the IRAM size? Please, include in your answer also the tools/commands that you've used to measure that.

> A more thorough analysis and refactoring of the code would be needed to make non-SSL builds use as little memory as before SSL was added to Sming.

You are welcome to provide fixes that improve this :)

gbl08ma, 2016-12-12:
I meant heap usage. It's possible IRAM usage has increased too but I didn't check. I did not check (nor do I particularly care) if the binary size (flash memory usage) has increased, but I suppose it has increased at least a little bit since now instantiating those classes requires a bit more work.

I'm working on these fixes as time allows.

slaff, 2016-12-29:
@gbl08ma Feel free to reopen this issue once you have your PR ready for review.

closed 2016-12-29 by slaff