SmingHub/Sming #831: ESP32-IDF compatability

slaff, 2016-12-12:
This issue is created to collect all the tasks we need to do in order to create a [ESP32-IDF](https://github.com/espressif/esp-idf) compatibility layer. 

We would like to estimate if it is feasible to migrate Sming and use it on ESP32 devices with the goals that we have set. 

```
If you have already dealt with code migration from ESP8266 to ESP32 
we would love to hear your experience as comments to this issue.
```

# Goals
[ ] Same code base. Adding feature/code changes to ESP8266 version should be easy to use directly in ESP32 or with minimal effort. The differences should come during the compilation and linking phase.

# Research
[ ] Migrate all changes from SmingRTOS that can be useful for ESP32 migration.
[ ] SDK functions have changed names.

closed 2017-03-11 by slaff

DjamesSuhanko, 2017-10-02:
Does anyone have news about sming for esp32? I can't develop to esp without sming framework anymore. :-)

slaff, 2020-05-06:
@DjamesSuhanko We have very-rough and very-early version support for an ESP32 via ESP-IDF version 4. You can follow the progress from here: https://github.com/slaff/Sming/tree/feature/arch-esp32

At the moment the only sample that compiles, links and can be flashed on a ESP32 microcontroller is the Basic_Blink example. I did not write "runs" because there is a lot of work to be done to reach this phase. Here are the steps that can help you test it.
```
# get the ESP32 branch version
cd /tmp
git -b feature/arch-esp32 checkout https://github.com/slaff/Sming.git
cd Sming
export SMING_HOME=$(pwd)/Sming

# now make sure to install ESP-IDF version 4.0
# Follow the instructions from here: 
# https://github.com/slaff/Sming/blob/feature/arch-
esp32/Sming/Arch/Esp32/README.rst#requirements

# once you have ESP-IDF v4.0 try to compile the Basic_Blink sample.
cd $SMING_HOME/../samples/Basic_Blink
make SMING_ARCH=Esp32

# once the compilation finishes you can flash the application to a device using
# the standard command

make flash
```

@mikee47 I would be very grateful if you can help us with the ESP32 architecture :) Take a look at the https://github.com/slaff/Sming/tree/feature/arch-esp32 branch. So far we need help for a lot of things. Let me just name the base ones:
- [ ] UART support
- [ ] Esp32 implementation of the different timers.
- [ ] Better GPIO support (at the moment only pinMode() and digitalWrite() should work )
- [ ] ADC support
- [ ] LWIP support.

Having this working will be a great start.

mikee47, 2020-05-07:
@slaff I'll try to find some time to get back into this, will definitely try out your branch! I can help by looking at UART support.

slaff, 2020-05-07:
> I'll try to find some time to get back into this,

@mikee47 This would be awesome! At the moment the Timer class(es) is what is probably with highest priority to run at least Basic_Blink. Up to know I was able to turn on and off leds but without the timer...

slaff, 2020-05-09:
> At the moment the Timer class(es) is what is probably with highest priority to run at least Basic_Blink.

As of now the hardware timer in Basic_Blink is working and the complete sample is running as expected. 

> I can help by looking at UART support.

Exactly what we need now. This is with highest priority and will help us run `Basic_Serial`.

slaff, 2020-05-14:
@mikee47 By the way we have pretty well working UART in Sming on ESP32 :). The code needs a bit of polishing/tuning but the main logic is there and it's working. I have opened a new ticket to track the progress: https://github.com/slaff/Sming/issues/57. The next task in line for me will be to see how to hook our Tcp and Udp routines in ESP-IDF. But it would be great if meanwhile someone can take a look at the i2s and i2c support :)