SmingHub/Sming #833: Problem building existing projects with latest git sources from develop branch

itarataiko, 2016-12-14:
Hi,
I've received latest git sources and can't compile my existing sources.
Can you help me?

/Volumes/case-sensitive/esp-open-sdk//sdk/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x354): undefined reference to `__wrap_system_restart_local'
/Volumes/case-sensitive/esp-open-sdk//sdk/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x435): undefined reference to `__wrap_system_restart_local'
/Volumes/case-sensitive/esp-open-sdk//sdk/lib/libpp.a(pp.o):(.text+0x30): undefined reference to `__wrap_system_restart_local'
/Volumes/case-sensitive/esp-open-sdk//sdk/lib/libpp.a(pp.o): In function `pp_soft_wdt_feed_local':
(.text+0x8c): undefined reference to `__wrap_system_restart_local'
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

slaff, 2016-12-15:
@itarataiko Try the following:

```
make -C $SMING_HOME clean
make -C $SMING_HOME

cd $SMING_HOME/../samples/Basic_Blink
make clean
make 
```

itarataiko, 2016-12-19:
thanks for your help

closed 2016-12-19 by itarataiko