SmingHub/Sming #840: SSL Build issue

GarageGadget, 2016-12-27:
Hi,

The Sming library builds successfully when using 'make all' however when building using the SSL option 'make ENABLE_SSL=1' the build fails.

I have added the requested third parties dependencies to the third-party directory.

OS = Windows 10
GCC = GCC V.5.1.0
Sming = V 3.0.0
Espressif= V1.5

**Build Output - Start**

make ENABLE_SSL=1
make[1]: Entering directory `/c/tools/sming/Sming/third-party/axtls-8266'
xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Wall -Os -g -O2 -v -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I/include -Icrypto -Issl -I.  -c -o crypto/hmac.o crypto/hmac.c
Using built-in specs.
COLLECT_GCC=c:\Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-gcc.exe
Target: xtensa-lx106-elf
Configured with: ../configure --prefix=/d/Neo/esp8266-devkit/Espressif/xtensa-lx106-elf --target=xtensa-lx106-elf --enable-multilib --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp=/d/Neo/esp8266-devkit/Espressif/build/gmp --with-mpfr=/d/Neo/esp8266-devkit/Espressif/build/mpfr --with-mpc=/d/Neo/esp8266-devkit/Espressif/build/mpc --enable-languages=c,c++ --with-newlib --disable-libssp --disable-__cxa_atexit
Thread model: single
gcc version 5.1.0 (GCC)
COLLECT_GCC_OPTIONS='-std=c99' '-D' 'ESP8266' '-Wall' '-Os' '-g' '-O2' '-v' '-Wpointer-arith' '-Wno-implicit-function-declaration' '-fno-inline-functions' '-nostdlib' '-mlongcalls' '-mno-text-section-literals' '-D' '__ets__' '-D' 'ICACHE_FLASH' '-I' 'C:/tools/mingw64/msys/1.0/include' '-I' 'crypto' '-I' 'ssl' '-I' '.' '-c' '-o' 'crypto/hmac.o'
 c:/espressif/xtensa-lx106-elf/bin/../libexec/gcc/xtensa-lx106-elf/5.1.0/cc1.exe -quiet -v -I C:/tools/mingw64/msys/1.0/include -I crypto -I ssl -I . -iprefix c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/ -D ESP8266 -D __ets__ -D ICACHE_FLASH crypto/hmac.c -quiet -dumpbase hmac.c -mlongcalls -mno-text-section-literals -auxbase-strip crypto/hmac.o -g -Os -O2 -Wall -Wpointer-arith -Wno-implicit-function-declaration -std=c99 -version -fno-inline-functions -o C:\Users\brent\AppData\Local\Temp\ccLdggXi.s
GNU C99 (GCC) version 5.1.0 (xtensa-lx106-elf)
        compiled by GNU C version 4.8.1, GMP version 6.0.0, MPFR version 3.1.2, MPC version 1.0.2
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/sys-include"
ignoring duplicate directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/include"
ignoring duplicate directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/include-fixed"
ignoring nonexistent directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/sys-include"
ignoring duplicate directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 C:/tools/mingw64/msys/1.0/include
 crypto
 ssl
 .
 c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/include
 c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/include-fixed
 c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/include
End of search list.
GNU C99 (GCC) version 5.1.0 (xtensa-lx106-elf)
        compiled by GNU C version 4.8.1, GMP version 6.0.0, MPFR version 3.1.2, MPC version 1.0.2
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 487d25fad2cff8b88e5853ec06c25fbf
In file included from c:\espressif\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\5.1.0\include\stdint.h:9:0,
                 from ssl/os_int.h:5,
                 from ssl/os_port.h:44,
                 from crypto/hmac.c:38:
c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\stdint.h:18:19: error: missing binary operator before token "("
 #if __GNUC_PREREQ (3, 2)
                   ^
In file included from ssl/os_port.h:63:0,
                 from crypto/hmac.c:38:
./util/time.h:6:3: error: unknown type name 'time_t'
   time_t tv_sec;
   ^
make[1]: *** [crypto/hmac.o] Error 1
make[1]: Leaving directory `/c/tools/sming/Sming/third-party/axtls-8266'
make: *** [compiler/lib/libaxtls.a] Error 2

**Build Output - Finished**

Your assistance is appreciated.

slaff, 2016-12-28:
@GarageGadget Can you tell me the values of these environment variables:
```
ESP_HOME
SMING_HOME
PATH
```

Which is the SDK that you use? Is it UDK, is it esp-alt-sdk? Is it self-built? 

Can you run the make command with `make ENABLE_SSL=1 V=1`  and paste the output here?

GarageGadget, 2016-12-28:
Environment Variables

 

C:\Windows\system32>echo %ESP_HOME%

c:\Espressif

 

C:\Windows\system32>echo %SMING_HOME%

C:\tools\sming\Sming

 

C:\Windows\system32>echo %path%

C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft SDKs\F#\4.0\Framework\v4.0\;C:\Program Files (x86)\MSBuild\14.0\bin;C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN;C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\Tools;C:\Windows\Microsoft.NET\Framework\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\VCPackages;C:\Program Files (x86)\HTML Help Workshop;C:\Program Files (x86)\Microsoft Visual Studio 14.0\Team Tools\Performance Tools;C:\Program Files (x86)\Windows Kits\10\bin\x86;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\;C:\ProgramData\Oracle\Java\javapath; C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\; C:\Program Files (x86)\Microsoft Emulator Manager\1.0\;C:\Program Files (x86)\nodejs\;C:\Program Files\Git\cmd;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\; C:\ProgramData\chocolatey\bin;C:\Program Files\Eclipse Foundation\4.4.2\eclipse;C:\tools\mingw64\bin;C:\tools\mingw64\msys\1.0\bin; c:\tools\sming;C:\Program Files (x86)\Nmap;C:\Espressif\xtensa-lx106-elf\bin

 

Currently UDK is being used

 

Make Output – Start

 

C:\tools\sming\Sming>make ENABLE_SSL=1 V=1

make -C third-party/axtls-8266 -e V=1 BIN_DIR=" /C/tools/sming/Sming/compiler/lib"

make[1]: Entering directory `/c/tools/sming/Sming/third-party/axtls-8266'

xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Wall -Os -g -O2 -v -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I/include -Icrypto -Issl -I.  -c -o crypto/bigint.o crypto/bigint.c

Using built-in specs.

COLLECT_GCC=c:\Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-gcc.exe

Target: xtensa-lx106-elf

Configured with: ../configure --prefix=/d/Neo/esp8266-devkit/Espressif/xtensa-lx106-elf --target=xtensa-lx106-elf --enable-multilib --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp=/d/Neo/esp8266-devkit/Espressif/build/gmp --with-mpfr=/d/Neo/esp8266-devkit/Espressif/build/mpfr --with-mpc=/d/Neo/esp8266-devkit/Espressif/build/mpc --enable-languages=c,c++ --with-newlib --disable-libssp --disable-__cxa_atexit

Thread model: single

gcc version 5.1.0 (GCC)

COLLECT_GCC_OPTIONS='-std=c99' '-D' 'ESP8266' '-Wall' '-Os' '-g' '-O2' '-v' '-Wpointer-arith' '-Wno-implicit-function-declaration' '-fno-inline-functions' '-nostdlib' '-mlongcalls' '-mno-text-section-literals' '-D' '__ets__' '-D' 'ICACHE_FLASH' '-I' 'C:/tools/mingw64/msys/1.0/include' '-I' 'crypto' '-I' 'ssl' '-I' '.' '-c' '-o' 'crypto/bigint.o'

c:/espressif/xtensa-lx106-elf/bin/../libexec/gcc/xtensa-lx106-elf/5.1.0/cc1.exe -quiet -v -I C:/tools/mingw64/msys/1.0/include -I crypto -I ssl -I . -iprefix c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/ -D ESP8266 -D __ets__ -D ICACHE_FLASH crypto/bigint.c -quiet -dumpbase bigint.c -mlongcalls -mno-text-section-literals -auxbase-strip crypto/bigint.o -g -Os -O2 -Wall -Wpointer-arith -Wno-implicit-function-declaration -std=c99 -version -fno-inline-functions -o C:\Users\brent\AppData\Local\Temp\ccI0RoXd.s

GNU C99 (GCC) version 5.1.0 (xtensa-lx106-elf)

        compiled by GNU C version 4.8.1, GMP version 6.0.0, MPFR version 3.1.2, MPC version 1.0.2

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072

ignoring nonexistent directory "c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/sys-include"

ignoring duplicate directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/include"

ignoring duplicate directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/include-fixed"

ignoring nonexistent directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/sys-include"

ignoring duplicate directory "c:/espressif/xtensa-lx106-elf/lib/gcc/../../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/include"

#include "..." search starts here:

#include <...> search starts here:

C:/tools/mingw64/msys/1.0/include

crypto

ssl

.

c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/include

c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/include-fixed

c:\espressif\xtensa-lx106-elf\bin\../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/include

End of search list.

GNU C99 (GCC) version 5.1.0 (xtensa-lx106-elf)

        compiled by GNU C version 4.8.1, GMP version 6.0.0, MPFR version 3.1.2, MPC version 1.0.2

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072

Compiler executable checksum: 487d25fad2cff8b88e5853ec06c25fbf

In file included from c:\espressif\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\5.1.0\include\stdint.h:9:0,

                 from ssl/os_int.h:5,

                 from ssl/os_port.h:44,

                 from crypto/bigint.c:68:

c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\stdint.h:18:19: error: missing binary operator before token "("

#if __GNUC_PREREQ (3, 2)

                   ^

make[1]: *** [crypto/bigint.o] Error 1

make[1]: Leaving directory `/c/tools/sming/Sming/third-party/axtls-8266'

make: *** [compiler/lib/libaxtls.a] Error 2

 

 

Make Output – End

slaff, 2016-12-28:
@GarageGadget I think I spotted the issue:

Under my test env I got the following:

```
xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Wall -Os -g -O2 -Wpointer-arith -Wno-im
plicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls
-mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I /c/Espressif/xtensa-lx10
6-elf//include -Icrypto -Issl -I.  -c -o crypto/bigint.o crypto/bigint.c
```

While you got the following

```
xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Wall -Os -g -O2 -v -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I/include -Icrypto -Issl -I.  -c -o crypto/bigint.o crypto/bigint.c
```

When I compared them carefully I saw that your include path is wrong: it is ` -I/include` when it should be `-I /c/Espressif/xtensa-lx10
6-elf//include`

Looking at the axtls makefile I can say that your `XTENSA_LIBS`  variable is not set correctly

```
CPPFLAGS += -I$(XTENSA_LIBS)/include \
		-Icrypto \
```

Can you do the following for me: 
* Open the axtls Makefile. 
* Replace these lines

```
XTENSA_LIBS ?= $(shell $(CC) -print-sysroot)
ifeq ($(XTENSA_LIBS),)
	XTENSA_LIBS = $(ESP_HOME)/xtensa-lx106-elf/
endif
```

With those 

```
XTENSA_LIBS ?= $(shell $(CC) -print-sysroot)

$(info "1:" $(XTENSA_LIBS))
ifeq ($(XTENSA_LIBS),)
	XTENSA_LIBS = $(ESP_HOME)/xtensa-lx106-elf/
        $(info "ESP_HOME:" $(ESP_HOME))
        $(info "2:" $(XTENSA_LIBS))
endif
```

Run again `make ENABLE_SSL=1 V=1` and paste the output.

GarageGadget, 2016-12-28:
Make file changes made as requests

 

C:\tools\sming\Sming>make ENABLE_SSL=1 V=1

mkdir -p out/build/system

mkdir -p out/build/system/helpers

mkdir -p out/build/Wiring

mkdir -p out/build/SmingCore

mkdir -p out/build/appinit

mkdir -p out/build/SmingCore/Network

mkdir -p out/build/SmingCore/Platform

mkdir -p out/build/Services/ArduinoJson

mkdir -p out/build/Services/CommandProcessing

mkdir -p out/build/Services/DateTime

mkdir -p out/build/Services/FATFS

mkdir -p out/build/Services/HexDump

mkdir -p out/build/Services/SpifFS

mkdir -p out/build/Services/WebHelpers

mkdir -p out/build/Services/cWebsocket

mkdir -p out/build/Services/libemqtt

mkdir -p out/build/Libraries/APA102

mkdir -p out/build/Libraries/Adafruit_GFX

mkdir -p out/build/Libraries/Adafruit_ILI9341

mkdir -p out/build/Libraries/Adafruit_NeoPixel

mkdir -p out/build/Libraries/Adafruit_PCD8544

mkdir -p out/build/Libraries/Adafruit_SSD1306

mkdir -p out/build/Libraries/Adafruit_ST7735

mkdir -p out/build/Libraries/ArduCAM

mkdir -p out/build/Libraries/BH1750FVI

mkdir -p out/build/Libraries/BMP180

mkdir -p out/build/Libraries/Bounce

mkdir -p out/build/Libraries/CapacitiveSensor

mkdir -p out/build/Libraries/DHT

mkdir -p out/build/Libraries/DS18S20

mkdir -p out/build/Libraries/DS3232RTC

mkdir -p out/build/Libraries/HMC5883L

[... 36 lines cut ...]
mkdir -p out/build/third-party/spiffs/src

mkdir -p out/build/gdb

mkdir -p out/build/third-party/axtls-8266/compat

mkdir -p out/build/third-party/axtls-8266/replacements

mkdir -p out/firmware

mkdir -p out/firmware/upgrade

/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -Isystem -Isystem/helpers -IWiring -ISmingCore -Iappinit -ISmingCore/Network/ -ISmingCore/Platform/ -IServices/ArduinoJson/ -IServices/CommandProcessing/ -IServices/DateTime/ -IServices/FATFS/ -IServices/HexDump/ -IServices/SpifFS/ -IServices/WebHelpers/ -IServices/cWebsocket/ -IServices/libemqtt/ -ILibraries/APA102/ -ILibraries/Adafruit_GFX/ -ILibraries/Adafruit_ILI9341/ -ILibraries/Adafruit_NeoPixel/ -ILibraries/Adafruit_PCD8544/ -ILibraries/Adafruit_SSD1306/ -ILibraries/Adafruit_ST7735/ -ILibraries/ArduCAM/ -ILibraries/BH1750FVI/ -ILibraries/BMP180/ -ILibraries/Bounce/ -ILibraries/CapacitiveSensor/ -ILibraries/DHT/ -ILibraries/DS18S20/ -ILibraries/DS3232RTC/ -ILibraries/HMC5883L/ -ILibraries/I2Cdev/ -ILibraries/IR/ -ILibraries/LiquidCrystal/ -ILibraries/MCP23017/ -ILibraries/MCP23S17/ -ILibraries/MMA_7455/ -ILibraries/Mirf/ -ILibraries/OneWire/ -ILibraries/RCSwitch/ -ILibraries/RF24/ -ILibraries/SDCard/ -ILibraries/SI7021/ -ILibraries/Servo/ -ILibraries/SparkFun_APDS9960/ -ILibraries/TFT_ILI9163C/ -ILibraries/Ultrasonic/ -ILibraries/WS2812/ -ILibraries/si4432/ -Ithird-party/spiffs/src -Igdb -Ithird-party/axtls-8266/compat -Ithird-party/axtls-8266/replacements -Isystem/include -Isystem/helpers/include -IWiring/include -ISmingCore/include -Iappinit/include -ISmingCore/Network//include -ISmingCore/Platform//include -IServices/ArduinoJson//include -IServices/CommandProcessing//include -IServices/DateTime//include -IServices/FATFS//include -IServices/HexDump//include -IServices/SpifFS//include -IServices/WebHelpers//include -IServices/cWebsocket//include -IServices/libemqtt//include -ILibraries/APA102//include -ILibraries/Adafruit_GFX//include -ILibraries/Adafruit_ILI9341//include -ILibraries/Adafruit_NeoPixel//include -ILibraries/Adafruit_PCD8544//include -ILibraries/Adafruit_SSD1306//include -ILibraries/Adafruit_ST7735//include -ILibraries/ArduCAM//include -ILibraries/BH1750FVI//include -ILibraries/BMP180//include -ILibraries/Bounce//include -ILibraries/CapacitiveSensor//include -ILibraries/DHT//include -ILibraries/DS18S20//include -ILibraries/DS3232RTC//include -ILibraries/HMC5883L//include -ILibraries/I2Cdev//include -ILibraries/IR//include -ILibraries/LiquidCrystal//include -ILibraries/MCP23017//include -ILibraries/MCP23S17//include -ILibraries/MMA_7455//include -ILibraries/Mirf//include -ILibraries/OneWire//include -ILibraries/RCSwitch//include -ILibraries/RF24//include -ILibraries/SDCard//include -ILibraries/SI7021//include -ILibraries/Servo//include -ILibraries/SparkFun_APDS9960//include -ILibraries/TFT_ILI9163C//include -ILibraries/Ultrasonic//include -ILibraries/WS2812//include -ILibraries/si4432//include -Ithird-party/spiffs/src/include -Igdb/include -Ithird-party/axtls-8266/compat/include -Ithird-party/axtls-8266/replacements/include -Iinclude -Isystem/include -IWiring -ILibraries -ISmingCore -I/c/Espressif/ESP8266_SDK/../include -Ithird-party/rboot -Ithird-party/rboot/appcode -Ithird-party/spiffs/src -Ithird-party/axtls-8266 -Iaxtls-8266/ssl -Ithird-party/axtls-8266/crypto -I /c/Espressif/ESP8266_SDK/include -I /c/Espressif/ESP8266_SDK/include/json -Wpointer-arith -Wundef -Werror -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE_FLASH -DARDUINO=106 -DCOM_SPEED_SERIAL= -Os -g -DLWIP_RAW=1 -DENABLE_SSL=1    -c system/irq_check.s -o out/build/system/irq_check.o

/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -Isystem -Isystem/helpers -IWiring -ISmingCore -Iappinit -ISmingCore/Network/ -ISmingCore/Platform/ -IServices/ArduinoJson/ -IServices/CommandProcessing/ -IServices/DateTime/ -IServices/FATFS/ -IServices/HexDump/ -IServices/SpifFS/ -IServices/WebHelpers/ -IServices/cWebsocket/ -IServices/libemqtt/ -ILibraries/APA102/ -ILibraries/Adafruit_GFX/ -ILibraries/Adafruit_ILI9341/ -ILibraries/Adafruit_NeoPixel/ - [... cut ...] -EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE_FLASH -DARDUINO=106 -DCOM_SPEED_SERIAL= -Os -g -DLWIP_RAW=1 -DENABLE_SSL=1    -c system/crash_handler.c -o out/build/system/crash_handler.o

In file included from system/crash_handler.c:24:0:

include/user_config.h:45:48: fatal error: ../axtls-8266/compat/lwipr_compat.h: No such file or directory

compilation terminated.

make: *** [out/build/system/crash_handler.o] Error 1

GarageGadget, 2016-12-28:
After further investigation I found that the lwipr_compat.h is not in the axtls-8266\compat folder. 

 

Further I downloaded lwirax from https://github.com/attachix/lwirax and copied the contents to axtls-8266\compat and re-built using make ENABLE_SSL=1 V=1 which progressed pass the lwirax dependencies.

However a new dependency is missing netdb.h referenced from os_port.h.

 

Please send me a list of dependencies that I can use to verify my installation of Sming for SSL.

 

Make Output Start

 

C:\tools\sming\Sming>make ENABLE_SSL=1 V=1

/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -Isystem -Isystem/helpers -IWiring -ISmingCore -Iappinit -ISmingCore/Network/ -ISmingCore/Platform/ -IServices/ArduinoJson/ -IServices/CommandProcessing/ -IServices/DateTime/ -IServices/FATFS/ -IServices/HexDump/ -IServices/SpifFS/ -IServices/WebHelpers/ -IServices/cWebsocket/ -IServices/libemqtt/ -ILibraries/APA102/ -ILibraries/Adafruit_GFX/ -ILibraries/Adafruit_ILI9341/ -ILibraries/Adafruit_NeoPixel/ -ILibraries/Adafruit_PCD8544/ -ILibraries/Adafruit_SSD1306/ -ILibraries/Adafruit_ST7735/ -ILibraries/ArduCAM/ -ILibraries/BH1750FVI/ -ILibraries/BMP180/ -ILibraries/Bounce/ -ILibraries/CapacitiveSensor/ -ILibraries/DHT/ -ILibraries/DS18S20/ -ILibraries/DS3232RTC/ -ILibraries/HMC5883L/ -ILibraries/I2Cdev/ -ILibraries/IR/ -ILibraries/LiquidCrystal/ -ILibraries/MCP23017/ -ILibraries/MCP23S17/ -ILibraries/MMA_7455/ -ILibraries/Mirf/ -ILibraries/OneWire/ -ILibraries/RCSwitch/ -ILibraries/RF24/ -ILibraries/SDCard/ -ILibraries/SI7021/ -ILibraries/Servo/ -ILibraries/SparkFun_APDS9960/ -ILibraries/TFT_ILI9163C/ -ILibraries/Ultrasonic/ -ILibraries/WS2812/ -ILibraries/si4432/ -Ithird-party/spiffs/src -Igdb -Ithird-party/axtls-8266/compat -Ithird-party/axtls-8266/replacements -Isystem/include -Isystem/helpers/include -IWiring/include -ISmingCore/include -Iappinit/include -ISmingCore/Network//include -ISmingCore/Platform//include -IServices/ArduinoJson//include -IServices/CommandProcessing//include -IServices/DateTime//include -IServices/FATFS//include -IServices/HexDump//include -IServices/SpifFS//include -IServices/WebHelpers//include -IServices/cWebsocket//include -IServices/libemqtt//include -ILibraries/APA102//include -ILibraries/Adafruit_GFX//include -ILibraries/Adafruit_ILI9341//include -ILibraries/Adafruit_NeoPixel//include -ILibraries/Adafruit_PCD8544//include -ILibraries/Adafruit_SSD1306//include -ILibraries/Adafruit_ST7735//include -ILibraries/ArduCAM//include -ILibraries/BH1750FVI//include -ILibraries/BMP180//include -ILibraries/Bounce//include -ILibraries/CapacitiveSensor//include -ILibraries/DHT//include -ILibraries/DS18S20//include -ILibraries/DS3232RTC//include -ILibraries/HMC5883L//include -ILibraries/I2Cdev//include -ILibraries/IR//include -ILibraries/LiquidCrystal//include -ILibraries/MCP23017//include -ILibraries/MCP23S17//include -ILibraries/MMA_7455//include -ILibraries/Mirf//include -ILibraries/OneWire//include -ILibraries/RCSwitch//include -ILibraries/RF24//include -ILibraries/SDCard//include -ILibraries/SI7021//include -ILibraries/Servo//include -ILibraries/SparkFun_APDS9960//include -ILibraries/TFT_ILI9163C//include -ILibraries/Ultrasonic//include -ILibraries/WS2812//include -ILibraries/si4432//include -Ithird-party/spiffs/src/include -Igdb/include -Ithird-party/axtls-8266/compat/include -Ithird-party/axtls-8266/replacements/include -Iinclude -Isystem/include -IWiring -ILibraries -ISmingCore -I/c/Espressif/ESP8266_SDK/../include -Ithird-party/rboot -Ithird-party/rboot/appcode -Ithird-party/spiffs/src -Ithird-party/axtls-8266 -Iaxtls-8266/ssl -Ithird-party/axtls-8266/crypto -I /c/Espressif/ESP8266_SDK/include -I /c/Espressif/ESP8266_SDK/include/json -Wpointer-arith -Wundef -Werror -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE_FLASH -DARDUINO=106 -DCOM_SPEED_SERIAL= -Os -g -DLWIP_RAW=1 -DENABLE_SSL=1    -c system/crash_handler.c -o out/build/system/crash_handler.o

In file included from third-party/axtls-8266/ssl/tls1.h:46:0,

                 from third-party/axtls-8266/ssl/ssl.h:75,

                 from third-party/rboot/../axtls-8266/compat/lwipr_compat.h:21,

                 from include/user_config.h:45,

                 from system/crash_handler.c:24:

third-party/axtls-8266/ssl/os_port.h:169:19: fatal error: netdb.h: No such file or directory

compilation terminated.

make: *** [out/build/system/crash_handler.o] Error 1

 

 

Make Output End

GarageGadget, 2016-12-28:
Further diagnostic testing shows that Sming "Tasty" branch builds successfully using "make ENABLE_SSL=1".

slaff, 2016-12-29:
I think I know what is happening. The patch for axtls is not applied because you already have the submodule downloaded. To prove my theory do the following:
* remove completely `Sming/third-party/axtls-8266` directory
* run `make ENABLE_SSL=1`

How did you get the submodules? How did you get the source code?

GarageGadget, 2016-12-29:
Make Output

 

C:\tools\sming\Sming>make ENABLE_SSL=1

Fetching third-party/axtls-8266/ ...

fatal: Not a git repository (or any of the parent directories): .git

make: *** [third-party/axtls-8266/Makefile] Error 128

 

How did you get the submodules? How did you get the source code?

 

The sub-modules with was retrieved from the external sources on github by downloading the zip files and using the ‘Without git repository” method of local copy installation.

slaff, 2016-12-29:
> The sub-modules with was retrieved from the external sources on github by downloading the zip files and using the ‘Without git repository” method of local copy installation.

@GarageGadget Understood. We will address this issue in the next release. Until then please get the latest source code using git. This can be done with the following command:

```
git clone https://github.com/SmingHub/Sming.git
```

You will need git version 1.9.x or newer to work properly. `Git` for windows can be downloaded from here: https://git-scm.com/download/win

GarageGadget, 2016-12-29:
Thanks for your assistance.

juppin, 2017-01-04:
Cloud it be that the folder in c:\tools is "Sming" and not "sming"?
In Makefile there is the standard path c:\tools\sming\Sming... 
That was my error!

closed 2017-01-11 by slaff