SmingHub/Sming #845: Sming 3.0.0 and SPIFFS image

STwilight, 2016-12-30:
After updating to Sming 3.0.0 when I building my project I've getting strange address for flashing SPIFFS image (out/firmware/spiff_rom.bin---------->0x50000).

If I'm flashing spiff_rom.bin to this address, module isn't starting, no any output in uart, it seems that module simply not booting.
If I'm flashing spiff_rom.bin to 0x4C000 (this address I was got on 2.1.0 Sming version), I'm getting module resetting by WDT and memory FatalExceptions in console.
If I'm flashing my module without SPIFFS image flashing, it seems, that it works well, but I need to push files into SPIFFS by FTP, that used in my project.

On Sming 2.1.0 everything was ok. Flashing address is 0x4C000 and it works correctly.

P.S. I'm using Win 7 x64 and UDK 2.0.9 by CHERTS (Espressif SDK 1.5.4). Module is ESP-01 with 512 kb of flash.

Any ideas, what it can be and in what way I can fix it? Thanks.

alonewolfx2, 2016-12-30:
Propably it's about firmware size and rom size is bigger than spi flash size or fw+spiffy is bigger than one rom size. Try to reduce spiffy size and flash normal address that's coming from spiffy

alonewolfx2, 2016-12-30:
@avr39-ripe i dont think so. i think issue is size.

hreintke, 2016-12-30:
There is an issue with calculating correct end-of-code size/start of SPIFFS in Sming NONOS.
Fixed with update in makefile & flashmem in SmingRTOS.

Might be causing the issue. You can check with comparing compile output & flash command

STwilight, 2016-12-31:
**Thank you all for the responses!**

I've updated my tools (NodeMCU flasher and ESP8266 Flashing Tool by Espressif) to the latest versions, and seems it works well.

@avr39-ripe, thank you for idea!
I'm planning to use rBoot at the future.

@alonewolfx2, please, describe, in what way I can reduce size of spiff image?
Here's sizes of my sections:
```
#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8374|     884
    rodata|           ReadOnly Data (RAM)|    3FFE8378|    3FFEAC5C|   10468
       bss|      Uninitialized Data (RAM)|    3FFEAC60|    3FFF1C70|   28688
      text|            Cached Code (IRAM)|    40100000|    40105E9A|   24218
irom0_text|           Uncached Code (SPI)|    40209000|    4024CA45|  277061
Total Used RAM : 40040
Free RAM : 41880
Free IRam : 8568
------------------------------------------------------------------------------
```
And SPIFFS info:
```
# Generating spiffs_bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
out/firmware/spiff_rom.bin---------->0x50000
```

@hreintke, are you talking about overall size in bytes of each section, that'll be flashed?
Here's ESP8266 flashing tool console output:
```
filename:  C:\tools\download tool\bin_tmp\downloadPanel1\0x00000.bin_rep
offset :  0
Writing 36864 @ 0x0...
Wrote 36864 bytes at 0x00000000 in 3.3 seconds (90.7 kbit/s)...

filename:  C:\tools\download tool\bin_tmp\downloadPanel1\0x09000.bin
offset :  36864
Writing 278528 @ 0x9000...
Wrote 278528 bytes at 0x00009000 in 34.1 seconds (65.3 kbit/s)...

filename:  C:\tools\download tool\bin_tmp\downloadPanel1\spiff_rom.bin
offset :  327680
Writing 196608 @ 0x50000...
Wrote 196608 bytes at 0x00050000 in 17.3 seconds (90.9 kbit/s)...
```

I'm not sure, that I'm fully understand addressing, so where I can read about it?

**Wishing good X-Mas and Happy new year to you all!**

slaff, 2017-02-09:
Feel free to reopen the issue if it is still not solved in the latest `develop` version.

closed 2017-02-09 by slaff