SmingHub/Sming #848: Sming 3.0 crash - Fatal exception (28)

fernandomorse, 2017-01-02:
After updating to Sming 3.0.0. I building my project and flash, but crash 

information:

make

#Memory / Section info:
------------------------------------------------------------------------------

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000374  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00002a04  3ffe8380  3ffe8380  00000460  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00006f68  3ffead88  3ffead88  00002e68  2**4
                  ALLOC
  3 .irom0.text   00042de8  40209000  40209000  000094a0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00006639  40100000  40100000  00002e64  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/build directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
Adding files in directory 'web/build'.
Added 'bootstrap.css.gz' to spiffs (15615 bytes).
Added 'style.css' to spiffs (7329 bytes).
Added 'wifi-sprites.png' to spiffs (1769 bytes).
Added 'index.html' to spiffs (13731 bytes).
Added 'settings.html' to spiffs (4093 bytes).
Added 'jquery.js.gz' to spiffs (30153 bytes).
out/firmware/spiff_rom.bin---------->0x4C000

flash

Killing Terminal to free /dev/ttyUSB0
/opt/Sming/Sming/Makefile-project.mk:399: recipe for target 'flash' failed
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x4C000 out/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0200

Wrote 40960 bytes at 0x0 in 3.6 seconds (92.0 kbit/s)...
Wrote 274432 bytes at 0x9000 in 23.8 seconds (92.2 kbit/s)...
Wrote 196608 bytes at 0x4c000 in 17.1 seconds (92.2 kbit/s)...

Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200
0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

any idea ?????

johndoe8967, 2017-01-03:
Hey,

seems to be a simple issue of memory overlap due to increased code size.
Just have a look at your memory map (difficult) or your flash log

Wrote 40960 bytes at 0x0 in 3.6 seconds (92.0 kbit/s)...
40960 decimal -> 0xa000 hex
Wrote 274432 bytes at 0x9000 in 23.8 seconds (92.2 kbit/s)...

This means that the second flash deleted something from the first flash because it started at 0x9000 so everything until 0xa000 is lost.

This usually happens with newer releases and I also had to optimise a lot.

hreintke, 2017-01-03:
@johndoe8967 
See my remark on SPIFFS in #845

fernandomorse, 2017-01-03:
ok. Iam solved using esp-open-sdk version 1.4.0. Version 1.5.4 is still bugged.

closed 2017-01-04 by slaff