SmingHub/Sming #849: Sming on Windows with UDK 2.2.1 / SDK 2.0

tius2000, 2017-01-03:
I know that SDK 2.0 is not supported yet, but I could not get the old version running so I tried in anyway. At least my simple applications do compile now after some trial-and-error.  Perhaps that might help someone else:

esptool2 seems to be no longer included in the UDK, but it is easy to get and compile:
- `git clone https://github.com/raburton/esptool2.git`
- `cd esptool2`
- `make`
- `mv esptool2.exe c:\Espressif\utils\ESP8266`

The utils path has changed, I needed to change in _%sming_home%\Makefile-windows.mk_:
- `SDK_TOOLS ?= $(ESP_HOME)/utils/ESP8266`

Autoscroll seems to be broken in the version of terminal.exe included in the UDK. I prefer plink.exe (part of the PuTTY package) anyway so I changed _%sming_home%\Makefile-windows.mk_:
- `KILL_TERM := taskkill -fi 'WINDOWTITLE eq \${COM_PORT}\'`
- `PLINK := d:/run/putty/plink.exe`
- `TERMINAL ?= start '"$(COM_PORT)"' $(PLINK) -serial $(COM_PORT) -sercfg 74880,8,n,1`

The newer SDK versions require the function _user_rf_cal_sector_set()_ to be supplied by the user. I simple added it to _%sming_home%\appinit\user_main.cpp_:

`extern "C" uint32 __attribute__((weak)) user_rf_cal_sector_set(void) {  see SDK sample code  }`

** THIS IS PROBABLY WRONG, I THINK THE CODE NEEDS TO BE ADAPTED BUT I DO NOT UNDERSTAND IMPLICATIONS YET - PERHAPS SOME WITH MORE INSIGHT CAN SUPPLY A CORRECT SOLUTION **

C files within the application create a linker error `undefined reference to '__ctype_ptr__'`. Changing the extension to .cpp seems to fix this.

Again, this was just blind trial-and-error :-)

see also #820

slaff, 2017-01-05:
> C files within the application create a linker error undefined reference to '__ctype_ptr__'. Changing the extension to .cpp seems to fix this.

@tius2000 That seems like a brutal hack to me.  Most probably some library linker flag is missing or not present by default. This will require further investigation ...

tius2000, 2017-01-09:
@slaff You are right, but I do not understand the build process good enough yet for this ;-)

In the meantime I made some changes to _Makefile-rbook.mk_ to implement the changes required by the new calibration sector:

    ...
    CFLAGS += -DRF_CAL_ADDR=$(RF_CAL_ADDR)
    ...
    flashinit:
    #   *** updated for SDK 2.0 (tiuw2000)
        $(vecho) "perform full flash erase and load default system parameters"
        $(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) erase_flash
        $(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) $(RF_CAL_ADDR) $(SDK_BASE)/bin/blank.bin 
        $(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) $(ESP_INIT_ADDR) $(SDK_BASE)/bin/esp_init_data_default.bin 
        $(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) $(ESP_PARM_ADDR) $(SDK_BASE)/bin/blank.bin 

The sectors are currently defined in my _Makefile-user.mk_, this might go to _Makefile-rbook.mk_ as well:

    ifeq ($(SPI_SIZE),4M)
        #   rf calibration sector
        RF_CAL_ADDR      = 0x3FB000

        #   system parameter area 
        ESP_INIT_ADDR    = 0x3FC000              
        ESP_PARM_ADDR    = 0x3FE000
    endif

_user_rf_cal_sector_set()_ becomes very simple now and can go to _application.cpp_:

    extern "C" uint32 __attribute__((weak)) user_rf_cal_sector_set(void) {
        return RF_CAL_ADDR >> 12;    
    }    

Again, this might not be the cleanest solution and should be revised by someone with more insight :-)

slaff, 2017-01-09:
@tius2000 Can you submit a PR with the changes and I will try to test it with esp-open-sdk and esp-alt-sdk on my system?

tius2000, 2017-01-10:
@slaff I'm still a noob regarding _git_ but I will try as soon as I get OTA working - seems to need some more tweaking ;-)

slaff, 2017-01-31:
@tius2000 Check this PR #921.

slaff, 2017-02-02:
@tius2000 Feel free to reopen this ticket if there are issues with using the latest `develop` with SDK 2.0

closed 2017-02-02 by slaff