SmingHub/Sming #863: Getting state of ESP when WDT reset occurs

brunodymios, 2017-01-07:
Hi there,
is it possible to retrieve the stack trace of the ESP when a **WDT reset** occured?

I can't get any information from the serial log about what's going on, such as **where the code gets stuck** for some reason.

**Additional question**: is it possibile in your opinion to have a WDT reset when accessing same method from different timer callbacks?

slaff, 2017-01-08:
> is it possible to retrieve the stack trace of the ESP when a WDT reset occured?

Yes, this is built-in in Sming with the crash handler. Once a WDT reset occurs you should see a stack dump that can be decoded with `tools/decode-stacktrace.py`

brunodymios, 2017-01-08:
@slaff unfortunately when a WDT occurs in my code I can't see the stacktrace but only something like this:

```
wdt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
```

The stacktrace is appearing only in case of fatal exceptions. What about investigating on which part of the code is causing the ESP to halt and hence the WDT to reset it?

slaff, 2017-01-09:
You can send me the code and I can compile and test it with the latest Sming and 1.5.x SDK.

brunodymios, 2017-01-09:
Hi @slaff, thank you.
It's not a matter of solving a specific problem on my code (already using latest Sming and SDK 1.5.x). It would be rather a best practice, something that may be added to the wiki, in order to help developers using Sming to debug such ESP8266 reboots (`ets Jan 8 2017,rst cause:4, boot mode:(3,7)`).

On the internet several people are experiencing random and pseudo-random reboots of the chip and I think that one of the most probable cause is related to something wrong in the code rather than power supply issues or something like that.
A (possibly easy) way to find out where the code got stuck causing the WDT to reboot the ESP, even without any fatal exception, would be a very helpful tool for developers.

Can anyone help on this?

Thanks

slaff, 2017-01-27:
Most of the times our crash handler will print a stack trace and point you to the place where the issue occurred.  But not always. For example:
*  When the WDT reset happens BEFORE our crash handler is loaded
*  When WDT reset happens due to interrupts that are conflicting with each other. And especially if that interrupt is related to printing/reading information from UART port.

>  random and pseudo-random ...
Hm... Sming has been relatively stable for us so random reboots mean either issue with your custom code OR the hardware that it works with. A WDT reset is most probably caused by software.  

Reset cause 4 is a normal software reset
```
REASON_SOFT_RESTART 	= 4,	/* software restart ,system_restart , GPIO status won’t change */
```
You may find something useful here: https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-messages-and-modes

closed 2017-01-27 by slaff