SmingHub/Sming #864: HttpServer* Examples and MeteoControl have Fatal exception (28)

bsilvereagle, 2017-01-07:
On a fresh pull of the `develop` branch (012f72b) I am able to build the HttpServer* and MeteoControl e xamples. However, upon flashing I immediately and continuously receive 

```
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
```

The exception fails to display after a reset, but the ESP does not appear to be doing anything. 

`0x4000228b` does not show up in `firmware.map` from a quick search.

slaff, 2017-01-08:
Erase your flash before saving the firmware on it.

```
./esptool.py --port /dev/ttyUSB0 erase_flash
```

If needed replace the port variable with your one.

bsilvereagle, 2017-01-08:
The issue still persists. I've attached the *.bin and *.map files. 

Unfortunately, my version of esp-open-sdk doesn't have the xtensa gdb, so I can't get any further. 

[out.zip](https://github.com/SmingHub/Sming/files/692122/out.zip)

slaff, 2017-01-09:
What is the flash size of your ESP?

bsilvereagle, 2017-01-09:
4MB, ESP201

IanAber, 2017-01-13:
I am having similar problems. As soon as I try any of the samples with HttpServer in them I get the looping Fatal Exception 28 error with the same addresses as above. I am running on a Mac and uploading to a NodeMCU board.

KarolBedkowski, 2017-01-19:
Hi,
Check ESP SDK version - esp-open-sdk by default install v2.0.0.
Version 1.5.4 works fine.

slaff, 2017-02-01:
@bsilvereagle Please, paste the output when you flash the ROMs on the device. Type `make flash V=1`.

bsilvereagle, 2017-02-04:
```
mkdir -p out/build/app
mkdir -p out/firmware
/opt/Esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -Iapp -Iapp/include -Iinclude -I/opt/Esp/Sming/Sming/include -I/opt/Esp/Sming/Sming/ -I/opt/Esp/Sming/Sming/system/include -I/opt/Esp/Sming/Sming/Wiring -I/opt/Esp/Sming/Sming/Libraries -I/opt/Esp/Sming/Sming/SmingCore -I/opt/Esp/Sming/Sming/Services/SpifFS -I/opt/Esp/esp-open-sdk/sdk/../include -I/opt/Esp/Sming/Sming/third-party/rboot -I/opt/Esp/Sming/Sming/third-party/rboot/appcode -I/opt/Esp/Sming/Sming/third-party/spiffs/src -I/opt/Esp/esp-open-sdk/sdk/include -Wpointer-arith -Wundef -Werror -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE_FLASH -DARDUINO=106 -DCOM_SPEED_SERIAL=115200  -Os -g -fno-rtti -fno-exceptions -std=c++11 -felide-constructors -c app/application.cpp -o out/build/app/application.o
/opt/Esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru out/build/app_app.a out/build/app/application.o
/opt/Esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -L/opt/Esp/Sming/Sming/compiler/lib/ -L/opt/Esp/esp-open-sdk/sdk/lib -T/opt/Esp/Sming/Sming/compiler/ld/eagle.app.v6.cpp.ld -nostdlib -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,-Map=out/firmware/firmware.map -Wl,-wrap,system_restart_local  -Wl,--start-group -lmicroc -lmicrogcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain -lsming -lcrypto -lpwm -lsmartconfig out/build/app_app.a -Wl,--end-group -o out/build/app.out
if [ -f "out/fwMeminfo" ]; then \
  mv out/fwMeminfo out/firmware/fwMeminfo.old; \
fi
/opt/Esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump -h -j .data -j .rodata -j .bss -j .text -j .irom0.text out/build/app.out > out/firmware/fwMeminfo.new
if [ -f "out/firmware/fwMeminfo.new" -a -f "out/firmware/fwMeminfo.old" ]; then \
  awk -F "|" 'FILENAME == "out/firmware/fwMeminfo.old" { arr[$1]=$5 } FILENAME == "out/firmware/fwMeminfo.new" { if (arr[$1] != $5){printf "%s%s%+d%s", substr($0, 1, length($0) - 1)," (",$5 - arr[$1],")\n" } else {print $0} }' out/firmware/fwMeminfo.old out/firmware/fwMeminfo.new; \
elif [ -f "out/firmware/fwMeminfo.new" ]; then \
  cat out/firmware/fwMeminfo.new; \
fi

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000374  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00002304  3ffe8380  3ffe8380  00000460  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00006f98  3ffea688  3ffea688  00002768  2**4
                  ALLOC
  3 .irom0.text   0003ee04  40209000  40209000  000098f0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         0000717f  40100000  40100000  00002764  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
# Generating spiffs_bin
if [ -d "files" ]; then \
	echo "files directory exists. Creating out/firmware/spiff_rom.bin"; \
	/opt/Esp/Sming/Sming/spiffy/spiffy 196608   files out/firmware/spiff_rom.bin; \
else \
	echo "No files found in ./files."; \
	echo "Creating empty out/firmware/spiff_rom.bin"; \
	/opt/Esp/Sming/Sming/spiffy/spiffy 196608   dummy.dir out/firmware/spiff_rom.bin; \
fi
No files found in ./files.
Creating empty out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
Creating empty filesystem.
pkill -9 -f "/dev/ttyUSB0 115200" || exit 0
/opt/Esp/Sming/Sming/Makefile-project.mk:399: recipe for target 'flash' failed
/opt/Esp/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x48000 out/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 40960 @ 0x0... 40960 (100 %)
Wrote 40960 bytes at 0x0 in 3.6 seconds (92.1 kbit/s)...
Writing 258048 @ 0x9000... 258048 (100 %)
Wrote 258048 bytes at 0x9000 in 22.4 seconds (92.3 kbit/s)...
Writing 196608 @ 0x48000... 196608 (100 %)
Wrote 196608 bytes at 0x48000 in 17.0 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200
depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
```

slaff, 2017-02-05:
> Writing 40960 @ 0x0... 40960 (100 %)
> Wrote 40960 bytes at 0x0 in 3.6 seconds (92.1 kbit/s)...
> Writing 258048 @ 0x9000... 258048 (100 %)

It seems that your first ROM is overlapping with the second one. Here is the math ( you can use python to test it):

start_address + size = 0x0 + 40960 = hex(0x0 + 40960) = 0xa000
your first ROM is ending at 0xa000 address. But the second one is starting at 0x9000, which is earlier:
```
0x9000 > 0xa000
False
```
and in fact your second ROM is deleting information from the first ROM due to overlapping. 
We already discussed what to do in situations like that in the [chat](https://gitter.im/SmingHub/Sming).

closed 2017-02-06 by slaff