SmingHub/Sming #866: What ESP8266 boards are compatible?

colombojrj, 2017-01-09:
First of all, thanks do the project. It is very well documented! 

But, there are several ESP8266 boards, for example, 12E, 01, etc. What are compatible or tested?

Thanks!

slaff, 2017-01-09:
@colombojrj Probably all ESPs with flash size >= 1MB will work. ESP-01 is having too little flash for most of the examples to work with the 1.5.x SDK.

Tested: I am using ESP-12E with 4MB flash for tests.

colombojrj, 2017-01-10:
Thanks man!

ADiea, 2017-01-12:
I used ESP 7 with 512K until recently now waiting for new hw based on 12E

johndoe8967, 2017-01-12:
I ordered a bunch of 512k from china and they work great for my projects.
Of course I've to check size limitations and I can't use all features in one project.

Ogyb, 2017-02-01:
i use ESP-7, ESP-02, ESP-201, ESP-07S (external antenna connector) for projects and ESP12 for testing.

anakod, 2017-02-11:
All boards compatible with Sming :)

closed 2017-02-11 by anakod