SmingHub/Sming #867: how to proceed after building the output

agauniyal, 2017-01-09:
I've successfully built the program which creates a new directory `out` containing `build` and `firmware` directories with files inside them. How should I proceed forward assuming I've all tools installed and esp8266 device connected on `/dev/ttyUSB0`?

agauniyal, 2017-01-09:
```
.
├── app
│   └── application.cpp
├── include
│   └── user_config.h
├── Makefile
├── Makefile-user.mk
└── out
    ├── build
    │   ├── app
    │   │   └── application.o
    │   ├── app_app.a
    │   └── app.out
    └── firmware
        ├── 0x00000.bin
        ├── 0x09000.bin
        ├── firmware.map
        └── fwMeminfo.new

6 directories, 11 files
```

slaff, 2017-01-09:
`make flash` should be the next command.

Take a look at the wiki page: https://github.com/SmingHub/Sming/wiki/Linux-Quickstart#build-and-flash-a-basic-blink-example-using-ttyusb0

closed 2017-01-11 by agauniyal