SmingHub/Sming #872: analogRead

fernandomorse, 2017-01-11:
I installed the Sming version 3.0 Windows. When trying to compile the mq-2 sensor program, I checked for error analogRead => error: 'analogRead' was not declared in this scope. I'm using system_adc_read (); , Is this the right solution ??
 
On the Linux platform it compiles without problems, using analogRead ().

slaff, 2017-01-12:
> error analogRead => error: 'analogRead' was not declared in this scope

analogRead is present. It is declared in [Sming/SmingCore/Digital.h](https://github.com/SmingHub/Sming/blob/4453de5e643cd6ec5e7a69b70c036c9e7e689c48/Sming/SmingCore/Digital.h). You might have forgotten to include it.

fernandomorse, 2017-01-12:
Dear Slaff,

analogRead is not present en Sming/SmingCore/Digital.h. Version Sming v3.0. Iam included programa lines.

Please close

Morse

slaff, 2017-01-13:
> analogRead is not present en Sming/SmingCore/Digital.h

@fernandomorse take a look at line 64: https://github.com/SmingHub/Sming/blob/4453de5e643cd6ec5e7a69b70c036c9e7e689c48/Sming/SmingCore/Digital.h#L64

closed 2017-01-13 by slaff