SmingHub/Sming #873: make of Basic_rBoot fails after updating Sming to 3.0.1

juppin, 2017-01-11:
Hello devs,

after i cloned the new Sming 3.0.1 to a new folder and set the paths in Makefile-user.mk of Basic_rBoot example, make fails with error in linker...

terminal output:

none@ubuntu:/opt/Sming/samples/Basic_rBoot$ make
OC out/build/libmain2.a
make -C /opt/Sming/Sming/third-party/rboot
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird betreten
CC rboot-stage2a.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot-stage2a.elf
E2 /opt/Sming/samples/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot.elf
E2 /opt/Sming/samples/Basic_rBoot/out/firmware/rboot.bin
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird verlassen
C+ app/application.cpp
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c
AR out/build/app_app.a
LD out/build/app_0.out
/opt/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find -lsming
collect2: error: ld returned 1 exit status
/opt/Sming/Sming/Makefile-rboot.mk:380: die Regel für Ziel „out/build/app_0.out“ scheiterte
make: *** [out/build/app_0.out] Fehler 1

```

slaff, 2017-01-11:
@juppin Did you compile the sming library before that?

juppin, 2017-01-11:
Yes.

This are the steps i have done:
1. Cloned the repo with `git clone https://github.com/SmingHub/Sming.git`.
2. Changed `ESP_HOME` and `SMING_HOME` of Makefile in Sming/Sming to my paths.
3. `make ENABLE_SSL=1`
4. cd to ../samples/Basic_rboot
5. I changed `ESP_HOME` and `SMING_HOME` of Makefile-user.mk in also to my paths.
6. `make`

The same error occured yesterday on my raspberry pi, now i have this done in my Ubuntu VM.
The same sample in my old directory with Sming 3.0.0 compiles without errors.

juppin, 2017-01-11:
Now i have cloned it a third time and compiled Sming first without `ENABLE_SSL=1` and tested the `Basic_rboot` example and it comiles sucsessful.

Then i do a `make clean` and a `make ENABLE_SSL=1` in SMING_HOME and the example works also with SSL enabled in SmingFramework.

Confusing...

Could there maybe a bug in the makefiles if one compiles Sming the first time after cloning with `ENABLE_SSL=1`?

slaff, 2017-01-12:
> Then i do a make clean and a make ENABLE_SSL=1 in SMING_HOME and the example works also with SSL enabled in SmingFramework.
> Confusing...

LibSmingSSL has a mechanism to be compiled automatically if it is requested. We should add the same mechanims to the nornal LibSming so that is less confusing. I will take care to add this for the next release.

PR #879 (2017-01-19): Compile libsming(ssl) on demand
Makes the build process more consistent and fixes #873.
commit eb3384b4 (2017-01-19): Compile libsming(ssl) on demand (#879)

Makes the build process more consistent and fixes #873.
closed 2017-01-19 by slaff