SmingHub/Sming #874: Access from Internet through Router fails

sndlover, 2017-01-13:
Hi everyone!
First of all: Thanks for this really great project, that got my ideas running very fast!

After implementing and testing my solution (based on the HTTPServer Examples) locally, which works like a charm, I wanted to have it accessible from outside through my internet DSL router. So I set up port forwarding correctly, but when i try to access my device from outside by my mobile phone or a friends computer, I only get some partial data. The index.html is delivered and small amounts of data like my Ajax requests and JSON data in return work, but larger chunks of data like an serving an image or JS files get stuck and time out. 

What I see on the debug output is, that per file to be sent, only 3 chunks of data are sent (pushcount=1, 2, 3) and then it times out, before having finished to send that file and steps over to the next file with the same result (I will post some serial logs later).

My router is a Fritzbox 7490 and i am connected to DSL16000.

I have absolutely no clue, what happens here, as the projects works perfectly inside my local network.
Probably this even is not an issue with SMING, but I would appreciate, if someone could give me a hint.

Btw: The same happens also with the original HTTPServer Examples.

Best Regards, Frank

avr39-ripe, 2017-01-13:
Known issue with esp nonos sdk + sming nonos.. Roots is in too huge mtu compiled in lwip lib. We need recompile lwip and reduce mtu. Not possible yet, but there is issue about integrating lwip as submodule, so we can recompile it and do not use binary one from espressif

sndlover, 2017-01-13:
I appreciate your quick answer very much and I am kind of glad to hear, that this is not only an exotic issue for me. 

Is there any workaround possible, until this is fixed by its root cause?

I assume, that I am not the only one, who wants to access their device from outside?
How do you others deal with it?

avr39-ripe, 2017-01-13:
Bad news here, you cat take old liblwip_535.a fro esp sdk 1.4.0 and try to change original liblwip.a in your esp sdk with it, but with such small mtu (haha.. one is too big and another one - too small :(( ) other problems arise, But give it a try, with smaller mtu I can access html pages from internet via router.
Another way is to use SmingRTOS which is based by esp rtos sdk - it can serve html rtc to internet out of the box.. but SmingRTOS has its own gotchas with stack etc... so NO SIMPLE WAY to solve it meanwhile.
See #515 for real solution, but not ready yet.

sndlover, 2017-01-13:
Thanks, i will give it a try with the older lib, when I am back home this evening.

Does it make any sense to play around with some defines in "c:\tools\Sming\Sming\system\include\lwip\opt.h"?

I see some MTU and fragmenting related defines there.
Does it have any effect, when I change them and rebuild SMING, or are those values already compiled into liblwip?

sndlover, 2017-01-13:
Hey Alexander,
I have to thank you for support, turning this issue from a total show stopper for my project  into something I can continue working with. Thank you very much!

What I did: I copied the liblwip356.a from SDK 1.4.1 over my current liblwip.a in SDK 1.5.0 and only rebuilt my app against the new (old) library. That got me a working web application, that functions also through my router. In the debug output I can see that the MTU (or is it MSS?) is lowered to the minimum of 536 bytes and more smaller packets are sent. But this far, I can not see any real disadvantage or instability.

Maybe it is possible to find these values in the lib and at least patch them, to find an optimum?
I'll give that a try.

Cheers, Frank

slaff, 2017-01-19:
@sndlover We have added LWIP source as an optional feature. It works with SDK 1.5. If you want to compile sming and use custom LWIP do the following:

```
cd $SMING_HOME
make clean
make ENABLE_CUSTOM_LWIP=1
```

If you do not need the `espconn` functionality then compile Sming in the following way:

```
cd $SMING_HOME
make clean
make ENABLE_CUSTOM_LWIP=1 ENABLE_ESPCONN=0
```

Make sure also to to compile your project with the ENABLE_CUSTOM_LWIP=1. For example
```
cd $SMING_HOME/../samples/HttpServer_AJAX
make clean
make ENABLE_CUSTOM_LWIP=1
```

If you want to always use the custom LWIP library in your project then add the option
```
ENABLE_CUSTOM_LWIP=1
```

in your `Makefile-user.mk` file.

If you are not sure which libraries are used then run make with `V=1` and check if the linking is done against `lwip_open` or `lwip`.

closed 2017-01-19 by slaff