SmingHub/Sming #876: printf: width and precision not working for string output?

tius2000, 2017-01-17:
I tried to make CommandHandler output look nicer. However, It seems to me that the width and precision specifiers for printf (e. g. %16s, %10.10s, %-10.10s) are ignored for strings.

ADiea, 2017-01-19:
Hi @tius2000 indeed %s does not support yet width specifiers
Please see \Sming\Sming\system\m_printf.cpp for refference

tius2000, 2017-01-20:
Thank you very much @ADiea! I tried to add support for width and precision to m_printf.cpp: 

    diff --git a/Sming/system/m_printf.cpp b/Sming/system/m_printf.cpp
    index 58ada9d..62b0da8 100644
    --- a/Sming/system/m_printf.cpp
    +++ b/Sming/system/m_printf.cpp
    @@ -19,6 +19,8 @@ void (*cbc_printchar)(char ch) = uart_tx_one_char;
     
     #define is_digit(c) ((c) >= '0' && (c) <= '9')
     
    +#define MIN(a, b)   ( (a) < (b) ? (a) : (b) )
    +
     static int skip_atoi(const char **s)
     {
        int i = 0;
    @@ -148,11 +150,14 @@ int m_vsnprintf(char *buf, size_t maxLen, const char *fmt, va_list args)
            width = 0;
            pad = ' ';
            base = 10;
    +        bool minus = 0;
     
            do
            {
    +            if ('-' == *fmt) minus = 1, fmt++;
    +            
                //skip width and flags data - not supported
    -			while ('+' == *fmt || '-' == *fmt || '#' == *fmt || '*' == *fmt || 'l' == *fmt)
    +			while ('+' == *fmt || '#' == *fmt || '*' == *fmt || 'l' == *fmt)
                    fmt++;
     
                if (is_digit(*fmt)) {
    @@ -179,20 +184,22 @@ int m_vsnprintf(char *buf, size_t maxLen, const char *fmt, va_list args)
                *str++ = (unsigned char) va_arg(args, int);
                continue;
     
    -		case 's':
    +		case 's': {
                s = va_arg(args, char *);
     
    -			if (!s)
    -			{
    -				s = "<NULL>";
    -			}
    -			else
    -			{
    -				while (*s && (maxLen - (uint32_t)(str - buf) > OVERFLOW_GUARD))
    -					*str++ = *s++;
    -			}
    +			if (!s) s = "(null)";
    +            size_t len = strlen(s);
    +            len     = MIN( len,   precision );
    +            len     = MIN( len,   maxLen - size_t(str - buf) - OVERFLOW_GUARD);
    +            width   = MIN( width, maxLen - size_t(str - buf) - OVERFLOW_GUARD);
    +
    +            int padding = width - len;
    +            while (!minus && padding-- > 0) *str++ = ' ';
    +            while (len--) *str++ = *s++;
    +            while (minus && padding-- > 0) *str++ = ' ';
     
                continue;
    +        }    
     
            case 'p':
                s = ultoa((unsigned long) va_arg(args, void *), tempNum, 16);

This small test seems to give correct results now:

    //  test width and precision for string output

    #include "stdio.h"
    #include "m_printf.h"

    static const char *formats[] = { 
        "%s",
        "%5s",
        "%-5s",
        "%5.5s",
        "%-5.5s",
        0
    };

    static const char *texts[] = {
        0,
        "123",
        "12345",
        "1234567"
    };

    typedef int (*P_PRINTF) (char const *str, ...);

    static void test( const char *title, P_PRINTF method ) {
        printf("\n%s\n\n", title);
        
        for (const char **pfmt = formats; *pfmt; pfmt++) {
            printf("    %-10.10s ", *pfmt);
            
            for (int j=0; j < sizeof(texts)/sizeof(const char *); j++) {
                printf("    '");
                method( *pfmt, texts[j]);
                printf("'");
            }        
            printf("\n");
        }
    }

    int main() {
        test("printf", printf);
        test("m_printf", m_printf);
    }

output (old):

    printf
        %s             '(null)'    '123'    '12345'    '1234567'
        %5s            '(null)'    '  123'    '12345'    '1234567'
        %-5s           '(null)'    '123  '    '12345'    '1234567'
        %5.5s          '(null'    '  123'    '12345'    '12345'
        %-5.5s         '(null'    '123  '    '12345'    '12345'

    m_printf
        %s             ''    '123'    '12345'    '1234567'
        %5s            ''    '123'    '12345'    '1234567'
        %-5s           ''    '123'    '12345'    '1234567'
        %5.5s          ''    '123'    '12345'    '1234567'
        %-5.5s         ''    '123'    '12345'    '1234567'
        
output (new):
 
    m_printf
        %s             '(null)'    '123'    '12345'    '1234567'
        %5s            '(null)'    '  123'    '12345'    '1234567'
        %-5s           '(null)'    '123  '    '12345'    '1234567'
        %5.5s          '(null'    '  123'    '12345'    '12345'
        %-5.5s         '(null'    '123  '    '12345'    '12345'

slaff, 2017-01-20:
@tius2000 Can you prepare a PR and @ADiea will review it?

tius2000, 2017-01-20:
PR #880, hope I got the right ;-)

closed 2017-01-23 by slaff