SmingHub/Sming #877: Sming as library to use in PlatformIO

chall3ng3r, 2017-01-17:
I've been looking for Sming support on PlatformIO and so far there's no solid development on it.

Is it possible to make Sming as library (compiled or src) to be just put in Lib folder of PlatformIO project targeting ESP and compile?

This will be easier than complicated build support for PlatformIO.

Matheus-Garbelini, 2017-01-18:
@slaff Yes, we need sming to be a selectable framework as soon as possible because sming should be a framework understandable to arduino users and not to just computers engineers. Currently not all people use Linux and installations on windows sometimes fails and its up to the user with "supposed" knowledge about linkers and environment setup to make it work. (A normal user wouldn't gess that in windows we need to download esptool2, and compile it, and fix the path in the configuration file, and manually change the flash and spiffs size).

Also eclipse is a little broken with the code completion and feels unnatural the way to compile in it. Putting sming in platformio would depreciate the current form of installation that is kinda unstable and allow non engineering personnel to be able to finally write the code and compile it without having to editing manually config files there and here. Not mentioning that we could use it in other IDEs, (platformio make system is 100 times easier to configure everything in other IDEs)

slaff, 2017-01-19:
We should analyze this one: https://github.com/platformio/platform-espressif8266/tree/feature/framework-sming and take a look at the steps described here: https://community.platformio.org/t/adding-a-new-framework-to-platformio/297/15.

If you guys are willing to invest some of your time and do the PlatformIO integration I might be able to help you if needed.

Matheus-Garbelini, 2017-01-21:
@slaff thank you. I'll get a look into this.

slaff, 2017-02-27:
Feel free to reopen the issue once there is some progress on it.

closed 2017-02-27 by slaff