SmingHub/Sming #883: WifiAccessPoint.config does not accept password setting.

sensorslink, 2017-01-21:
WifiAccessPoint.config does not accept password setting:

C+ app/application.cpp
app/application.cpp: In function 'void init()':
app/application.cpp:855:38: error: no matching function for call to 'AccessPointClass::config(const char [5], const char [5])'
  WifiAccessPoint.config("user","pass");

When I use :  WifiAccessPoint.config("use,"pass",AUTH_OPEN);  is accepted...

Checking the SmingCore / Platform / AccessPoint.h and AccessPoint.cpp , Using the AUTH_OPEN parameter the password is aborted ...

Snippet of code:AccessPoint.cpp 

bool AccessPointClass::config(String ssid, String password, AUTH_MODE mode, bool hidden /* = false*/, int channel /* = 7*/, int beaconInterval /* = 200*/)
{
softap_config config = {0};

	if (mode == AUTH_WEP) return false; // Not supported!

	if (mode == AUTH_OPEN)
		password = "";
        bool enabled = isEnabled();
        enable(true);
        wifi_softap_dhcps_stop();
        wifi_softap_get_config(&config);
        config.channel = channel;
        config.ssid_hidden = hidden;
        memset(config.ssid, 0, sizeof(config.ssid));
        memset(config.password, 0, sizeof(config.password));
        strcpy((char*)config.ssid, ssid.c_str());
        strcpy((char*)config.password, password.c_str());
        config.ssid_len = ssid.length();
        config.authmode = mode;
        config.max_connection = 4; // max 4
        config.beacon_interval = beaconInterval;
        ...

Someone is experiencing this problem or has a solution, need to have password in AP mode.

avr39-ripe, 2017-01-21:
Hello, @sensorslink, AUTH_OPEN mean no password, use next line as proper example, change two first parameters to ssid and password and keep AUTH_WPA2_PSK.
WifiAccessPoint.config(WIFIAP_SSID, WIFIAP_PWD, AUTH_WPA2_PSK);

sensorslink, 2017-01-21:
I made the change in the code and it works fine

thank you.

closed 2017-01-21 by sensorslink