SmingHub/Sming #887: Patches for axtls and ummalloc fail

ADiea, 2017-01-23:
Hi,
Building on Windows with UDK [v2.2.1]
Clean sming clone, develop branch.

Looks like axtls patch cannot be applied because the search string is not found. Is this working on windows? Could it be some parameter for git apply patch to ignore whitespaces etc? 

Is anyone still building on windows/udk can you confirm sming building OK with ENABLE_SSL=1

Output
'''
19:47:09 **** Build of configuration Build (GNU) for project sming ****
make rebuild DISABLE_SPIFFS=1 ENABLE_SSL=1 
rm -f out/build/app_app.a
rm -f out/build/app.out
rm -rf out/build/system out/build/system/helpers out/build/Wiring out/build/SmingCore out/build/appinit out/build/SmingCore/Network/ out/build/SmingCore/Platform/ out/build/Services/ArduinoJson/ out/build/Services/CommandProcessing/ out/build/Services/DateTime/ out/build/Services/FATFS/ out/build/Services/HexDump/ out/build/Services/SpifFS/ out/build/Services/WebHelpers/ out/build/Services/cWebsocket/ out/build/Services/libemqtt/ out/build/Libraries/APA102/ out/build/Libraries/Adafruit_GFX/ out/build/Libraries/Adafruit_ILI9341/ out/build/Libraries/Adafruit_NeoPixel/ out/build/Libraries/Adafruit_PCD8544/ out/build/Libraries/Adafruit_SSD1306/ out/build/Libraries/Adafruit_ST7735/ out/build/Libraries/ArduCAM/ out/build/Libraries/BH1750FVI/ out/build/Libraries/BMP180/ out/build/Libraries/Bounce/ out/build/Libraries/CapacitiveSensor/ out/build/Libraries/DHT/ out/build/Libraries/DS18S20/ out/build/Libraries/DS3232RTC/ out/build/Libraries/HMC5883L/ out/build/Libraries/I2Cdev/ out/build/Libraries/IR/ out/build/Libraries/LiquidCrystal/ out/build/Libraries/MCP23017/ out/build/Libraries/MCP23S17/ out/build/Libraries/MMA_7455/ out/build/Libraries/Mirf/ out/build/Libraries/OneWire/ out/build/Libraries/RCSwitch/ out/build/Libraries/RF24/ out/build/Libraries/SDCard/ out/build/Libraries/SI7021/ out/build/Libraries/Servo/ out/build/Libraries/SparkFun_APDS9960/ out/build/Libraries/TFT_ILI9163C/ out/build/Libraries/Ultrasonic/ out/build/Libraries/WS2812/ out/build/Libraries/si4432/ out/build/third-party/spiffs/src out/build/gdb out/build/third-party/axtls-8266/compat out/build/third-party/axtls-8266/replacements
rm -rf out/build
rm -rf out/firmware
make --no-print-directory -C third-party/axtls-8266 clean -e V=1 BIN_DIR=" /D/prj/homa/SmingTest/sming/Sming/compiler/lib"
make[1]: *** No rule to make target `clean'.  Stop.
make: [clean] Error 2 (ignored)
git submodule update --init --recursive third-party/axtls-8266/
Submodule 'third-party/axtls-8266' (https://github.com/igrr/axtls-8266.git) registered for path 'Sming/Sming/third-party/axtls-8266'
Cloning into 'D:/prj/homa/SmingTest/sming/Sming/third-party/axtls-8266'...
Submodule path 'third-party/axtls-8266': checked out 'd768568ae799ad41c2e6668ac54d7896584ad667'
Submodule 'compat' (https://github.com/attachix/lwirax.git) registered for path 'third-party/axtls-8266/compat'
Cloning into 'D:/prj/homa/SmingTest/sming/Sming/third-party/axtls-8266/compat'...
Submodule path 'third-party/axtls-8266/compat': checked out 'bc2bf7b61b9295b07d2fd9b5a9bb7824b1677c85'
touch third-party/.patches/axtls-8266.patch
cd third-party/axtls-8266/; git apply -v  /D/prj/homa/SmingTest/sming/Sming/third-party/.patches/axtls-8266.patch
d:/prj/homa/SmingTest/sming/Sming/third-party/.patches/axtls-8266.patch:5: trailing whitespace.
/*
d:/prj/homa/SmingTest/sming/Sming/third-party/.patches/axtls-8266.patch:6: trailing whitespace.
 libc_replacements.c - replaces libc functions with functions
d:/prj/homa/SmingTest/sming/Sming/third-party/.patches/axtls-8266.patch:7: trailing whitespace.
 from Espressif SDK
d:/prj/homa/SmingTest/sming/Sming/third-party/.patches/axtls-8266.patch:8: trailing whitespace.
 Copyright (c) 2015 Ivan Grokhotkov. All rights reserved.
d:/prj/homa/SmingTest/sming/Sming/third-party/.patches/axtls-8266.patch:9: trailing whitespace.
 This file is part of the esp8266 core for Arduino environment.
Checking patch replacements/libc.c...
Checking patch replacements/mem.c...
Checking patch Makefile...
error: while searching for:
MFORCE32 := $(shell $(CC) --help=target | grep mforce-l32)

XTENSA_LIBS ?= $(shell $(CC) -print-sysroot)

OBJ_FILES := \
	crypto/aes.o \

error: patch failed: Makefile:7
error: Makefile: patch does not apply
Checking patch ssl/os_port.h...
error: while searching for:

#if defined(ESP8266)

#include "util/time.h"
#include <errno.h>
#define alloca(size) __builtin_alloca(size)
#define TTY_FLUSH()

error: patch failed: ssl/os_port.h:60
error: ssl/os_port.h: patch does not apply
Checking patch ssl/tls1.c...
error: while searching for:
    int read_len, is_client = IS_SET_SSL_FLAG(SSL_IS_CLIENT);
    uint8_t *buf = ssl->bm_data;

    if (IS_SET_SSL_FLAG(SSL_SENT_CLOSE_NOTIFY))
        return SSL_CLOSE_NOTIFY;

error: patch failed: ssl/tls1.c:1334
error: ssl/tls1.c: patch does not apply
Checking patch ssl/tls1.h...
error: while searching for:
#include "version.h"
#include "config.h"
#include "os_int.h"
#include "os_port.h"
#include "crypto.h"
#include "crypto_misc.h"

error: patch failed: ssl/tls1.h:43
error: ssl/tls1.h: patch does not apply
Checking patch tools/make_certs.sh...
error: while searching for:
# THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
#

AXDIR=`pwd`/`dirname $0`
CWD=`mktemp -d` && cd $dir
cd $CWD 

error: patch failed: tools/make_certs.sh:29
error: tools/make_certs.sh: patch does not apply
make -C  /D/prj/homa/SmingTest/sming/Sming
make: [third-party/axtls-8266/Makefile] Error 1 (ignored)
make[1]: Entering directory `/D/prj/homa/SmingTest/sming/Sming'
mkdir -p out/build/system
mkdir -p out/build/system/helpers
mkdir -p out/build/Wiring
mkdir -p out/build/SmingCore
mkdir -p out/build/appinit
mkdir -p out/build/SmingCore/Network
mkdir -p out/build/SmingCore/Platform
mkdir -p out/build/Services/ArduinoJson
mkdir -p out/build/Services/CommandProcessing
mkdir -p out/build/Services/DateTime
mkdir -p out/build/Services/FATFS
mkdir -p out/build/Services/HexDump
mkdir -p out/build/Services/SpifFS
mkdir -p out/build/Services/WebHelpers
mkdir -p out/build/Services/cWebsocket
mkdir -p out/build/Services/libemqtt
mkdir -p out/build/Libraries/APA102
mkdir -p out/build/Libraries/Adafruit_GFX
mkdir -p out/build/Libraries/Adafruit_ILI9341
mkdir -p out/build/Libraries/Adafruit_NeoPixel
mkdir -p out/build/Libraries/Adafruit_PCD8544
mkdir -p out/build/Libraries/Adafruit_SSD1306
mkdir -p out/build/Libraries/Adafruit_ST7735
mkdir -p out/build/Libraries/ArduCAM
mkdir -p out/build/Libraries/BH1750FVI
mkdir -p out/build/Libraries/BMP180
mkdir -p out/build/Libraries/Bounce
mkdir -p out/build/Libraries/CapacitiveSensor
mkdir -p out/build/Libraries/DHT
mkdir -p out/build/Libraries/DS18S20
mkdir -p out/build/Libraries/DS3232RTC
mkdir -p out/build/Libraries/HMC5883L
mkdir -p out/build/Libraries/I2Cdev
mkdir -p out/build/Libraries/IR
mkdir -p out/build/Libraries/LiquidCrystal
mkdir -p out/build/Libraries/MCP23017
mkdir -p out/build/Libraries/MCP23S17
mkdir -p out/build/Libraries/MMA_7455
mkdir -p out/build/Libraries/Mirf
mkdir -p out/build/Libraries/OneWire
mkdir -p out/build/Libraries/RCSwitch
mkdir -p out/build/Libraries/RF24
mkdir -p out/build/Libraries/SDCard
mkdir -p out/build/Libraries/SI7021
mkdir -p out/build/Libraries/Servo
mkdir -p out/build/Libraries/SparkFun_APDS9960
mkdir -p out/build/Libraries/TFT_ILI9163C
mkdir -p out/build/Libraries/Ultrasonic
mkdir -p out/build/Libraries/WS2812
mkdir -p out/build/Libraries/si4432
mkdir -p out/build/third-party/spiffs/src
mkdir -p out/build/gdb
mkdir -p out/build/third-party/axtls-8266/compat
mkdir -p out/build/third-party/axtls-8266/replacements
mkdir -p out/firmware
mkdir -p out/firmware/upgrade
make -C third-party/axtls-8266 -e V=1 BIN_DIR=" /D/prj/homa/SmingTest/sming/Sming/compiler/lib"
make[2]: Entering directory `/D/prj/homa/SmingTest/sming/Sming/third-party/axtls-8266'
xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Wall -Os -g -O2 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I/include -Icrypto -Issl -I.  -c -o crypto/aes.o crypto/aes.c
In file included from c:\espressif\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\5.2.0\include\stdint.h:9:0,
                 from ssl/os_int.h:5,
                 from ssl/os_port.h:44,
                 from crypto/aes.c:38:
c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\stdint.h:18:19: error: missing binary operator before token "("
 #if __GNUC_PREREQ (3, 2)
                   ^
In file included from ssl/os_port.h:63:0,
                 from crypto/aes.c:38:
./util/time.h:6:3: error: unknown type name 'time_t'
   time_t tv_sec;
   ^
make[2]: *** [crypto/aes.o] Error 1
make[2]: Leaving directory `/D/prj/homa/SmingTest/sming/Sming/third-party/axtls-8266'
make[1]: *** [compiler/lib/libaxtls.a] Error 2
make[1]: Leaving directory `/D/prj/homa/SmingTest/sming/Sming'
make: *** [third-party/axtls-8266/Makefile] Error 2

19:47:22 Build Finished (took 13s.406ms)

'''

ADiea, 2017-01-23:
it looks like git apply looks for 
```
<SPACE>MFORCE32 := $(shell $(CC) --hel...
```
but in the Makefile the string is without leading space. Looking for way to ignore the leading space in the patches

ADiea, 2017-01-23:
No I think that the error is beacause git searches for "line1 \n line2" and my sources are cloned using windows style "line1 \r\n line2" -> so this is probably from a setting on my side, will test and close the issue but should be present as a warn on getting started page

ADiea, 2017-01-24:
Still having issues here applying the patch. It says it does not find a section of 7 lines at the begining of axtls/src/Makefile starting from line 7 but I look in the Makefile and the section is there...anyone had this issue before? I also started a bash console with Administrator rights fearing that it cannot read Makefile same error, still investigating. Line endings and trailing spaces seem not to be related to this

slaff, 2017-01-24:
Just for the test try to apply the patch manually with the `patch` command instead of `git apply ..` and tell me if it works for you.

ADiea, 2017-01-24:
It finally worked with --ignore-whitespace
```
--ignore-whitespace

    When applying a patch, ignore changes in whitespace in context lines if necessary. Context lines will preserve their whitespace, and they will not undergo whitespace fixing regardless of the value of the --whitespace option. New lines will still be fixed, though.
```
The thing is I compared the Makefile with the patch hunk and there were no differences so this is really strange.

using patch command or using git apply --reject applied libc.c and mem.c patches in /replacements but still failed on Makefile and subsequent files. I will make again a clean clone and test with modified makefile with whitespace option included and if it works will submit a PR. Strange noone elese has this issue

ADiea, 2017-01-24:
Fixed by setting global CRLF config for windows. This should be added to appropiate getting started section for windows
```
git config --global core.autocrlf input
```
now patches apply cleanly
```
Checking patch replacements/libc.c...
Checking patch replacements/mem.c...
Checking patch Makefile...
Checking patch ssl/os_port.h...
Checking patch ssl/tls1.c...
Checking patch ssl/tls1.h...
Checking patch tools/make_certs.sh...
Applied patch replacements/libc.c cleanly.
Applied patch replacements/mem.c cleanly.
Applied patch Makefile cleanly.
Applied patch ssl/os_port.h cleanly.
Applied patch ssl/tls1.c cleanly.
Applied patch ssl/tls1.h cleanly.
Applied patch tools/make_certs.sh cleanly.
warning: 2 lines add whitespace errors.
```

closed 2017-01-24 by ADiea

slaff, 2017-01-24:
> This should be added to appropiate getting started section for windows

@ADiea Can you please do this?

ADiea, 2017-01-24:
Done, put it in wiki -> getting started for windows page.

slaff, 2017-01-24:
> Done, put it in wiki -> getting started for windows page.

@ADiea Great! Thanks a lot.