SmingHub/Sming #889: iram overflowed when ENABLE_SSL=1

lxkarthi, 2017-01-24:
Whenever ENABLED_SSL is enabled in my codes, the iram overflows.

```
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -L/opt/Sming/Sming/compiler/lib/ -L/opt/esp-open-sdk/sdk/lib -Lout/build -Trom0.ld -nostdlib -u call_user_start -u Cache_Read_Enable_New -Wl,-static -Wl,--gc-sections -Wl,-Map=out/build/app_0.map -Wl,-wrap,system_restart_local  -Wl,--start-group out/build/app_app.a -lmicroc -lmicrogcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain2 -lsmingssl -lcrypto -lpwm -lsmartconfig -laxtls -Wl,--end-group -o out/build/app_0.out
/opt/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: out/build/app_0.out section `.text' will not fit in region `iram1_0_seg'
/opt/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: region `iram1_0_seg' overflowed by 35495 bytes
collect2: error: ld returned 1 exit status
make: *** [out/build/app_0.out] Error 1
```

I noticed in the linker script this line in irom0.text
`  *libsming.a:*(.literal .text .literal.* .text.* .stub .gnu.warning .gnu.linkonce.literal.* .gnu.linkonce.t.*.literal .gnu.linkonce.t.*)
`
When SSL is enabled, instead of libsming.a, libsmingssl.a is used as shown in above linker command (-lsmingssl)
The following line should be added in irom0.text section for consistency.
`  *libsmingssl.a:*(.literal .text .literal.* .text.* .stub .gnu.warning .gnu.linkonce.literal.* .gnu.linkonce.t.*.literal .gnu.linkonce.t.*)
`.

slaff, 2017-01-24:
It's present since quite some time in here https://github.com/SmingHub/Sming/blob/develop/Sming/compiler/ld/eagle.app.v6.cpp.ld#L164 or here https://github.com/SmingHub/Sming/blob/develop/samples/MqttClient_Hello/rom0.ld#L164 . 

Which of the samples is failing? Or it is just your app? If we are talking about the latter then it is your own responsibility to add it in your application.

lxkarthi, 2017-01-24:
my rom0.ld is older. It from SMING version before SSL is added. Thanks.
The differences are libsmingssl.a and 
  irom0_0_seg :                         org = 0x40202010, len = (1M - 0x2010)
Is 1M dependent on the what ESP board we are using?
I am using ESP-12.
By the way, How do it convert SSL cerificate.pem.crt and private.pem.key to HEX array representation in ssl/cert.h and ssl/private_key.h as in MqttClient_Hello sample?

slaff, 2017-01-24:
> Is 1M dependent on the what ESP board we are using?

It depends on the size of the flash memory chip. Max is 1M but if you have smaller flash size you should adjust this accordingly. If you have flash which is bigger than 1M do not change the value because ESP8266 can address up to 1M. 

> How do it convert SSL cerificate.pem.crt

With `xxd`. Take a look at the end of that script: https://github.com/igrr/axtls-8266/blob/d768568ae799ad41c2e6668ac54d7896584ad667/tools/make_certs.sh#L179

closed 2017-01-24 by slaff