SmingHub/Sming #891: make problem  commit  be2b257

kwis2, 2017-01-24:
I have downloaded commit sming be2b257 into fresh PC with win 7. I have tried to make and got: 

> C:\tools\sming\Sming>make
> AS system/irq_check.s
> AS gdb/gdbstub-entry.s
> CC system/crash_handler.c
> CC system/flashmem.c
> CC Services/FATFS/ff.c
> CC Services/SpifFS/spiffs_sming.c
> In file included from Services/SpifFS/spiffs_sming.h:8:0,
>                  from Services/SpifFS/spiffs_sming.c:1:
> third-party/spiffs/src/spiffs.h:300:5: error: "SPIFFS_OBJ_META_LEN" is not defin
> ed [-Werror=undef]
>  #if SPIFFS_OBJ_META_LEN
>      ^
> third-party/spiffs/src/spiffs.h:311:5: error: "SPIFFS_OBJ_META_LEN" is not defin
> ed [-Werror=undef]
>  #if SPIFFS_OBJ_META_LEN
>      ^
> In file included from Services/SpifFS/spiffs_sming.h:8:0,
>                  from Services/SpifFS/spiffs_sming.c:1:
> third-party/spiffs/src/spiffs.h:534:5: error: "SPIFFS_OBJ_META_LEN" is not defin
> ed [-Werror=undef]
>  #if SPIFFS_OBJ_META_LEN
>      ^
> cc1.exe: all warnings being treated as errors
> make: *** [out/build/Services/SpifFS//spiffs_sming.o] Error 1
> 
> C:\tools\sming\Sming>

Maybe it is a problem with new commit of spiffs ?

kwis2, 2017-01-24:
With old commit of spiffs (0.3.5) make Sming succesfull. With current spiffs version 0.3.6 Sming fail

slaff, 2017-01-27:
@kwis2 Either you or your git is misbehaving.  We are using older state of spiffy where this variable is not present (it started to appear 8 days ago: https://github.com/pellepl/spiffs/blame/5d4d1a48840fa00ff61c4b847dfcfed9b1123260/src/default/spiffs_config.h#L120).  In our code the submodule should come from: spiffs @ 21fe570 (https://github.com/pellepl/spiffs/tree/21fe570974103f71cf7ff02f9888c77746fbc66d).

Most probable cause is that you have decided to update also the submodules to their latest versions, which is not recommended.

closed 2017-01-27 by slaff

kwis2, 2017-01-27:
@slaff  In win 7 I use manual install with git from .gitmodules and there is `url = https://github.com/pellepl/spiffs.git` which link to newest commit