SmingHub/Sming #900: axTLS vs espconn_secure

icodk, 2017-01-26:
Currently using Sming with espconn_secure for communicating with https servers. Is there any advantage using the newly added axTLS  in terms of RAM speed, functionality etc.
With axTLS  I can check crt.  fingerprints Can it also be done with espconn_secure ?

slaff, 2017-01-26:
> With axTLS I can check crt. fingerprints Can it also be done with espconn_secure ?

It looks like you've answered your question. Plus with axTLS you can use TLS extensions(set SNI, set Max Fragment Size), you get better signing algorithms (weak once are removed), you get the latest source code that can be analyzed, more options to optimize the memory usage, better randomness and other small or big things that are important for security.

icodk, 2017-01-26:
@slaff Thanks
Important points. Not obvious for me before you mentioned it

closed 2017-01-26 by slaff