SmingHub/Sming #906: HardwareTimer and ESP8266_new_pwm seem to crash spiffs access

tius2000, 2017-01-29:
Whenever _HardwareTimer_ or _ESP8266_new_pwm_ is active, spiffs (e. g. _HttpResponse::sendFile_) becomes instable on my system and leads to a reboot. Perhaps this is a symptom of a hardware conflict or an unsafe interrupt operation. Has anyone experienced the same problem?

If this is confirmed and the underlying cause cannot be solved, perhaps a safe guard around the spiffs calls with ETS_FRC1_INTR_DISABLE() should be implemented within Sming?

slaff, 2017-01-30:
@tius2000 Please, ask also [spiffs](https://github.com/pellepl/spiffs/issues) and [ESP8266_new_pwm](https://github.com/StefanBruens/ESP8266_new_pwm/issues) communities.

slaff, 2017-02-02:
@tius2000 Feel free to reopen the ticket once you have more information from one of the other projects.

closed 2017-02-02 by slaff