SmingHub/Sming #914: Better understanding of fatal exception  at epc1=0x4000228b

slaff, 2017-01-30:
This issue is created to help us collect and understand the reason(s) for fatal exception at epc1=0x4000228b.

Reported problems: #857, #864.

If you have an idea what can cause this issue, please comment below. Hints for fixes, workarounds, etc are welcomed.

anakod, 2017-01-30:
Is ESP power enough? It can be reason for strange errors with reading flash memory.

slaff, 2017-01-30:
@anakod Do you have an idea where this error is coming from? What is the reason for seeing such an error?

anakod, 2017-01-31:
I'm not sure, just have error like that once, but I didn't remember exact epc1 code so it can be different error also. But I think first what users should to do in cases like that - check power source :) If voltage drops on some operations - it's probably reason.

lex77, 2017-01-31:
I have similar problem too. This fatal exception appears with sample projects (e.g. MeteoControl, Basic_WebSkeletonApp) and my own projects, that compiled with Sming v3.0.1 (esp-open-sdk-1.5.4).
But the same code compiled with Sming v2.1.5 works correct.

My test hardware is NodeMCU DEV board. I'm not sure that the problem is the lack of power. I will continue testing.

anakod, 2017-01-31:
If error depends on Sming version then it shouldn't be related to hardware\power source.
Error repeats every time or just in some cases?

ADiea, 2017-01-31:
+can we get additional info from crash handler / stack?

slaff, 2017-01-31:
@ADiea my guess is that the issue is happening even before our application is loaded/started. I have seen this when the first level bootloader is not able to check/load the ROM file. An error message like this is shown:

```
380, len 12528, room 8
tail 8
chksum 0x7c
csum 0x7c
csum err  <--------------------- Notice: We have checksum error reported.
ets_main.c Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28):
```

ADiea, 2017-01-31:
could it be caused by faulty flash chip (reached end of life) ?

mtzfactory, 2017-01-31:
Hi,

I'm having the same issue with my code, based on MQTT and HttpServer_ConfigNetwork:

```
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

```
The problem, I think, is related with the size of the first bin file: 0x0000.bin, that is bigger than the expected by the configuration in the make file:

**Makefile-project.mk**
`$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x09000 $(FW_BASE)/0x09000.bin $(SPIFF_START_OFFSET) $(SPIFF_BIN_OUT)`

It spect to be less than 36864 bytes (0x9000 ), but with the new Sming version is 40960 bytes (0xA000).

Sming 3.0 (esp-open-sdk 1.5.4 or 2.0.0) -- **NOT WORKING**
```
Flash params set to 0x0200
Writing 40960 @ 0x0...

Wrote 40960 bytes at 0x0 in 3.6 seconds (91.5 kbit/s)...
Writing 270336 @ 0x9000...

Wrote 270336 bytes at 0x9000 in 23.4 seconds (92.6 kbit/s)...
Writing 196608 @ 0x4c000...

Wrote 196608 bytes at 0x4c000 in 17.0 seconds (92.5 kbit/s)...
Leaving...
```

Sming 2.0 --> **WORKING**
```
make flash 
Killing Terminal to free /dev/cu.wchusbserial1a1230
/opt/esp-open-sdk/esptool/esptool.py -p /dev/cu.wchusbserial1a1230 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x4C000 out/firmware/spiff_rom.bin
Connecting...
Erasing flash...
Took 0.68s to erase flash block

Writing at 0x00000000... (2 %)
...
Writing at 0x00008800... (100 %) 
Wrote 35840 bytes at 0x00000000 in 4.2 seconds (68.2 kbit/s)...
Erasing flash...
Took 1.44s to erase flash block

Writing at 0x00009000... (0 %)
...
Writing at 0x00048c00... (100 %) 
Wrote 262144 bytes at 0x00009000 in 30.9 seconds (67.9 kbit/s)...
Erasing flash...
Took 1.04s to erase flash block

Writing at 0x0004c000... (0 %)
...
Writing at 0x0007bc00... (100 %) 
Wrote 196608 bytes at 0x0004c000 in 22.6 seconds (69.5 kbit/s)...
```

Any idea? solution?

mtzfactory, 2017-01-31:
BTW, can someone tell where these magical numbers come from? Thanks!

slaff, 2017-01-31:
@mtzfactory Take a look at the [commits her](https://github.com/SmingHub/Sming/pull/921). Meanwhile I will try to push a patch for esptool.py that gives warning or stops with an error if there is an overlap detected.

mtzfactory, 2017-02-01:
Hi @slaff, I just follow your [commits in the PR](https://github.com/SmingHub/Sming/pull/921) and didn't worked, I tried with different examples and even with the Basic_Blink example didn't succeed, when I connect to the serial port it prints some kind of random values.

PD: To be sure I did it right, I cloned again the Sming repo and applied your instructions there... but same result...

slaff, 2017-02-01:
@anakod Thanks for noticing this. I guess I made a mistake somewhere... Have to check...

slaff, 2017-02-02:
> Basic_Blink example didn't succeed, when I connect to the serial port it prints some kind of random values

@mtzfactory That is exactly what it does. It does not print any additional information on the serial and the only thing that should happen is your LED blinking. If you want to test an app that prints something in the console use our Basic_Serial example.

mtzfactory, 2017-02-02:
Thanks! but I already knew that @slaff... I just modified it with a Serial.println every time blink function was called, just to test everything was correct, but it wasn't... As the other examples, it prints some kind of random values, I think is reseting itself after a while...

anakod, 2017-02-03:
> I'm using a NodeMCU v0.9, what I'm missing?
You have only one board? Will be very good to test on different hardware also to be sure.

mtzfactory, 2017-02-14:
Hi @anakod ,

Yes I have: an esp-01 (1Mbyte flash)... and I tried with the new sming version: 3.1.0.
These are the results of the test:

Sming 2.0.0, esp-01 (1Mbyte flash), **Basic_Blink** works perfectly.
Sming 2.0.0, NodeMCU v0.9, **Basic_Blink** works perfectly.

Sming 3.1.0, esp-01 (1Mbyte flash), **Basic_Blink** works perfectly.
Sming 3.1.0, NodeMCU v0.9, **Basic_Blink** works perfectly.

Sming 2.0.0, esp-01 (1Mbyte flash), **HttpServer_ConfigNetwork** works perfectly.
Sming 2.0.0, NodeMCU v0.9, **HttpServer_ConfigNetwork** works perfectly.

Sming 3.1.0, esp-01 (1Mbyte flash), **HttpServer_ConfigNetwork doesn't work**.
Sming 3.1.0, NodeMCU v0.9, **HttpServer_ConfigNetwork doesn't work**.

In **both cases** I get the following message:

```timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 900
fs.start: size:3143880 Kb, offset:0x4024a000

***** Fatal exception 0
pc=0x40246e45 sp=0x3ffff8c0 excvaddr=0x00000000
ps=0x00000030 sar=0x0000001e vpri=0x65736666
r00: 0x40246e3c=1076129340 r01: 0x3ffff8c0=1073739968 r02: 0x3ffea90e=1073654030 
r03: 0x00000001=         1 r04: 0x00000100=       256 r05: 0x00eaa101=  15376641 
r06: 0x60000200=1610613248 r07: 0x00020000=    131072 r08: 0x00e8a101=  15245569 
r09: 0xfffdffff=   -131073 r10: 0x0000000c=        12 r11: 0x0000000c=        12 
r12: 0x4024a000=1076142080 r13: 0x00000004=         4 r14: 0x00000004=         4 
r15: 0x00000000=         0 

Stack dump:
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
================================================================
3ffff8c0:  40246e45 00000030 0000001e 65736666  
3ffff8d0:  40246e3c 3ffea90e 00000001 00000100  
3ffff8e0:  00eaa101 60000200 00020000 00e8a101  
3ffff8f0:  fffdffff 0000000c 0000000c 4024a000  
3ffff900:  00000004 00000004 00000000 00000000  
3ffff910:  0000000a 3ffe8370 00000000 40239d95  
3ffff920:  76726553 74207265 6f656d69 75207475  
3ffff930:  74616470 3a676e69 20303420 39203e2d  
3ffff940:  0a0d3030 ffffff00 3ffeaf60 00000000  
3ffff950:  0000000a 401004e2 0000000a ffffffff  
3ffff960:  40001da0 0000000a 0000001a ffffffff  
3ffff970:  40001db4 00000064 00000000 0000000a  
3ffff980:  40001f46 0000001c 60000200 3ffeaf70  
3ffff990:  40004b31 3ffffa20 00000004 4024a000  
3ffff9a0:  40101656 00dd9999 00000005 ffffff80  
3ffff9b0:  40246e3c 4024a000 00000004 00000004  
3ffff9c0:  3fff1ac8 3ffffa20 00000004 40246f82  
3ffff9d0:  3ffff9f0 3ffff9e0 00000004 3fff196c  
3ffff9e0:  4024a000 00000004 4024a000 3ffffa70  
3ffff9f0:  3ffffa70 00000002 0000ff00 00000000  
3ffffa00:  3ffeac40 402358f0 3ffffa50 4023b55c  
3ffffa10:  000203e9 3ffeae08 00000000 00000028  
3ffffa20:  40249fff 3ffeae08 00000028 40246d84  
3ffffa30:  3fff0b30 00000000 3fff1bb0 3ffeae08  
3ffffa40:  3ffeac40 402358f0 3fff1928 4023b5e1  
3ffffa50:  4023b394 4023b3b4 4023b358 bfe32000  
3ffffa60:  4024a000 00001000 00002000 00000100  
3ffffa70:  00000000 00000001 00000002 40237ec8  
3ffffa80:  00000006 00000009 0000000c 0000000d  
3ffffa90:  0000000e 0000000f 00000014 0000001c  
[... 61 lines cut ...]
3ffffe70:  00000000 00000000 ffffffff ffffffff  
3ffffe80:  ffffffff ffffffff ffffffff ffffffff  
3ffffe90:  ffffffff ffffffff ffffffff ffffffff  
3ffffea0:  ffffffff ffffffff ffffffff ffffffff  
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffec0:  ffffffff ffffffff ffffffff ffffffff  
3ffffed0:  ffffffff ffffffff ffffffff ffffffff  
3ffffee0:  ffffffff ffffffff ffffffff ffffffff  
3ffffef0:  ffffffff ffffffff ffffffff ffffffff  
3fffff00:  ffffffff ffffffff ffffffff ffffffff  
3fffff10:  ffffffff ffffffff ffffffff ffffffff  
3fffff20:  ffffffff ffffffff ffffffff ffffffff  
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff ffffffff ffffffff ffffffff  
3fffff60:  ffffffff ffffffff ffffffff ffffffff  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff 00000000 00000000 00000000  
3fffff90:  ffff0000 00000000 00000000 51000000  
3fffffa0:  000203e9 40004928 ffffff01 55aa55aa  

================================================================
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.```

slaff, 2017-02-14:
Please, paste the decoded stack trace here.
```
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
```

mtzfactory, 2017-02-15:
Hi @slaff ,

Here you have:

```
 ✘ mtz@mtzBookPro  /Volumes/Sming/Sming/samples/HttpServer_ConfigNetwork  ➦ f26c0f4 ● 
 python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
3ffff8c0:  40246e45 00000030 0000001e 65736666  
3ffff8d0:  40246e3c 3ffea90e 00000001 00000100  
3ffff8e0:  00eaa101 60000200 00020000 00e8a101  
3ffff8f0:  fffdffff 0000000c 0000000c 4024a000  
3ffff900:  00000004 00000004 00000000 00000000  
3ffff910:  0000000a 3ffe8370 00000000 40239d95  
3ffff920:  76726553 74207265 6f656d69 75207475  
3ffff930:  74616470 3a676e69 20303420 39203e2d  
3ffff940:  0a0d3030 ffffff00 3ffeaf60 00000000  
3ffff950:  0000000a 401004e2 0000000a ffffffff  
3ffff960:  40001da0 0000000a 0000001a ffffffff  
3ffff970:  40001db4 00000064 00000000 0000000a  
3ffff980:  40001f46 0000001c 60000200 3ffeaf70  
3ffff990:  40004b31 3ffffa20 00000004 4024a000  
3ffff9a0:  40101656 00dd9999 00000005 ffffff80  
3ffff9b0:  40246e3c 4024a000 00000004 00000004  
3ffff9c0:  3fff1ac8 3ffffa20 00000004 40246f82  
3ffff9d0:  3ffff9f0 3ffff9e0 00000004 3fff196c  
3ffff9e0:  4024a000 00000004 4024a000 3ffffa70  
3ffff9f0:  3ffffa70 00000002 0000ff00 00000000  
3ffffa00:  3ffeac40 40230x40246e45: flashmem_read_internal at /Volumes/Sming/Sming/Sming/system/flashmem.c:204
0x40246e3c: flashmem_read_internal at /Volumes/Sming/Sming/Sming/system/flashmem.c:200
0x4024a000: ?? ??:0
0x40239d95: m_vprintf at /Volumes/Sming/Sming/Sming/system/m_printf.cpp:86
0x401004e2: wdt_feed at ??:?
0x40001da0: ?? ??:0
0x40001db4: ?? ??:0
0x40001f46: ?? ??:0
0x40004b31: ?? ??:0
0x4024a000: ?? ??:0
0x40101656: spi_flash_read at ??:?
0x40246e3c: flashmem_read_internal at /Volumes/Sming/Sming/Sming/system/flashmem.c:200
0x4024a000: ?? ??:0
0x40246f82: flashmem_read at /Volumes/Sming/Sming/Sming/system/flashmem.c:83
0x4024a000: ?? ??:0
0x4024a000: ?? ??:0
b55c  
3ffffa10:  000203e9 3ffeae08 00000000 00000028  
3ffffa20:  40249fff 3ffeae08 00000028 40246d84  
3ffffa30:  3fff0b30 00000000 3fff1bb0 3ffeae08  
3ffffa40:  3ffeac40 402358f0 3fff1928 4023b5e1  
3ffffa50:  4023b394 4023b3b4 4023b358 bfe32000  
3ffffa60:  4024a000 00001000 00002000 00000100  
3ffffa70:  00000000 00000001 00000002 40237ec8  
3ffffa80:  00000006 00000009 0000000c 0000000d  
3ffffa90:  0000000e 0000000f 00000014 0000001c  
3ffffaa0:  0000001d 402358f0 3ffeaa40 402398b0  
3ffffab0:  4020a364 402358f0 3fff1928 402392af  
3ffffac0:  4020ab22 4020ab1d 3fff1928 4020ab25  
3ffffad0:  3fff1928 00001000 3ffeac62 4010076e  
3ffffae0:  0000007d 60000200 000000ed 00000000  
3ffffaf0:  00000000 3fff19d0 00000000 ffffffff  
3ffffb00:  ffffffff ffffffff ffff0002 00000000  
3ffffb10:  00000000 00000000 00000000 00000000  
3ffffb20:  00000000 00000000 00000000 00000000  
3ffffb30:  ffffffff 00ffffff 00000000 00000000  
3ffffb40:  00000000 00000000 00000000 00000x402358f0: __divsf3 at ??:?
0x4023b55c: spiffs_mount_internal at /Volumes/Sming/Sming/Sming/Services/SpifFS/spiffs_sming.c:158
0x40249fff: ?? ??:0
0x40246d84: flashmem_get_size_sectors at /Volumes/Sming/Sming/Sming/system/flashmem.c:150
0x402358f0: __divsf3 at ??:?
0x4023b5e1: spiffs_mount at /Volumes/Sming/Sming/Sming/Services/SpifFS/spiffs_sming.c:158
0x4023b394: api_spiffs_read at /Volumes/Sming/Sming/Sming/Services/SpifFS/spiffs_sming.c:158
[... 60 lines cut ...]
3ffffe20:  ffffffff ffffffff ffffffff ffffffff  
3ffffe30:  ffffffff ffffffff ffffffff ffffffff  
3ffffe40:  ffffffff 00000003 ffffffff ffffffff  
3ffffe50:  ffffffff ffffffff ffffffff ffffffff  
3ffffe60:  ffffffff ffffffff ffffffff ffff00c8  
3ffffe70:  00000000 00000000 ffffffff ffffffff  
3ffffe80:  ffffffff ffffffff ffffffff ffffffff  
3ffffe90:  ffffffff ffffffff ffffffff ffffffff  
3ffffea0:  ffffffff ffffffff ffffffff ffffffff  
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffec0:  ffffffff ffffffff ffffffff ffffffff  
3ffffed0:  ffffffff ffffffff ffffffff ffffffff  
3ffffee0:  ffffffff ffffffff ffffffff ffffffff  
3ffffef0:  ffffffff ffffffff ffffffff ffffffff  
3fffff00:  ffffffff ffffffff ffffffff ffffffff  
3fffff10:  ffffffff ffffffff ffffffff ffffffff  
3fffff20:  ffffffff ffffffff ffffffff ffffffff  
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff ffffffff ffffffff ffffffff  
3fffff60:  ffffffff ffffffff ffffffff ffffffff  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff 00000000 00000000 00000000  
3fffff90:  ffff0000 00000000 00000000 51000000  
3fffffa0:  000203e9 40004928 ffffff01 55aa55aa

slaff, 2017-02-15:
@mtzfactory Thanks. First make sure to have SPI_SIZE set to the correct size of the flashmem (what is your ESP device flashmen size by the way?). For example if you have 4 megabytes then set it to `SPI_SIZE=4M`. After that get the latest develop version, and then do the following:

```
cd $SMING_HOME
make dist-clean
cd ../samples/HttpServer_ConfigNetwork
make 
make flash
```

Test and write back with the results that you get.

ADiea, 2017-02-15:
looks like it crashes when trying to mount spiff

mtzfactory, 2017-02-16:
Hi @slaff,

With the develop version it works perfectly... Yeey!!!

Thanks a lot!

slaff, 2017-02-16:
The main cause for the fatal exception epc1=0x4..228b was overlapping ROMs. We proposed fix to estool.py team and it is now merged in the upstream version.  Thanks to all who helped us track the problem.

closed 2017-02-16 by slaff