SmingHub/Sming #922: Sming 3.0.1 MqttClient_Hello enabled spiffs error in fileGetContent

itarataiko, 2017-02-01:
Hi,

I try to add file system interaction to the MqttClient_Hello sample.
1. added files folder with some file with content
2. in the init() I added spiffs_mount()
3. after system ready try to receive file content with fileGetContent(filename)
4. received error read errno -10004 and Serial.println(fileGetContent(filename)) show me file name on the spiffs but not the file content

Thank for your help

slaff, 2017-02-01:
@itarataiko Did you comment out the following line from the makefile (https://github.com/SmingHub/Sming/blob/develop/samples/MqttClient_Hello/Makefile-user.mk#L37) ?

itarataiko, 2017-02-01:
yes I did

itarataiko, 2017-02-01:
and added 
SPIFF_FILES = files

itarataiko, 2017-02-01:
sorry, it was a mistake when issue was closed

itarataiko, 2017-02-02:
@slaff Did you reproduce this issue?

itarataiko, 2017-02-02:
I have fixed this issue by change spiffs_mount() with spiffs_mount_manual(0x40300000,SPIFF_SIZE)

closed 2017-02-02 by itarataiko

slaff, 2017-02-02:
> I have fixed this issue by change spiffs_mount() with spiffs_mount_manual(0x40300000,SPIFF_SIZE)

@itarataiko There is a new PR coming with changes related to spiffs mounting. You should use in the future `spiffs_mount_manual(<flash-address>,SPIFF_SIZE)` instead of `spiffs_mount_manual(<memory-mapped-address>,SPIFF_SIZE)` .  This means that if you save your spiffs at address 0x100000 on the flash memory you should also call `spiffs_mount_manual(0x100000,SPIFF_SIZE)` in your code. And that is more intuitive than what we have now.

itarataiko, 2017-02-02:
thanks for your detailed answer