SmingHub/Sming #930: Develop version of Sming still don't support SSL?

TomaszKn, 2017-02-05:
It's true?
When I'm try compile samples/Basic_Ssl I get:
`CC /home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming//third-party/rboot/appcode/rboot-api.c
AR out/build/app_app.a
LD out/build/app_0.out
out/build/libmain2.a(time.o): In function `millis':
(.irom0.text+0x94): multiple definition of `millis'
/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming/third-party/axtls-8266/replacements/time.c:33: first defined here
out/build/libmain2.a(time.o): In function `micros':
(.irom0.text+0xd0): multiple definition of `micros'
/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
out/build/libmain2.a(time.o): In function `configTime':
(.irom0.text+0x138): multiple definition of `configTime'
/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming/third-party/axtls-8266/replacements/time.c:34: first defined here
out/build/libmain2.a(time.o): In function `time':
(.irom0.text+0x1dc): multiple definition of `time'
/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/home/tomek/elektronika/esp8266/SDK2.0/Sming/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
collect2: error: ld returned 1 exit status
`

slaff, 2017-02-05:
@TomaszKn Please, do the following:
```
cd $SMING_HOME
git checkout develop
git pull
rm compiler/lib/libsming*.a
make clean
make ENABLE_SSL=1 
```

and report if there are compiler issues.

TomaszKn, 2017-02-05:
`cd $SMING_HOME

tomek@Lx64:xx/Sming/Sming$ git checkout develop
Already on 'develop'
Your branch is up-to-date with 'origin/develop'.
tomek@Lx64:xx/Sming/Sming$ git pull
Already up-to-date.
tomek@Lx64:xx/Sming/Sming$ rm compiler/lib/libsming*.a
tomek@Lx64:xx/Sming/Sming$ make clean
tomek@Lx64:xx/Sming/Sming$ make ENABLE_SSL=1 `

`AS system/irq_check.s
AS gdb/gdbstub-entry.s
CC system/flashmem.c
CC system/crash_handler.c
.
.
.
C+ Libraries/CapacitiveSensor/CapacitiveSensor.cpp

AR out/build/app_app.a
Installing libsmingssl
 SSL support is enabled
Done
Making spiffy utility
CC spiffy.c
CC ../third-party/spiffs/src/spiffs_cache.c
CC ../third-party/spiffs/src/spiffs_nucleus.c
CC ../third-party/spiffs/src/spiffs_hydrogen.c
CC ../third-party/spiffs/src/spiffs_gc.c
CC ../third-party/spiffs/src/spiffs_check.c
LD spiffy
Done
`
 then I try compile:

> tomek@Lx64:Xx/Sming/samples/Basic_Ssl$ make clean
tomek@Lx64:xx/Sming/samples/Basic_Ssl$ make ENABLE_SSL=1 
OC out/build/libmain2.a
make -C /xx/Sming/Sming//third-party/rboot RBOOT_GPIO_ENABLED=0
make[1]: Wejście do katalogu 'xx/Sming/Sming/third-party/rboot'
CC rboot-stage2a.c
LD /xx/Sming/samples/Basic_Ssl/out/build/rboot-stage2a.elf
E2 /xx/Sming/samples/Basic_Ssl/out/build/rboot-hex2a.h
CC rboot.c
LD /xx/Sming/samples/Basic_Ssl/out/build/rboot.elf
E2 /xx/Sming/samples/Basic_Ssl/out/firmware/rboot.bin
make[1]: Opuszczenie katalogu 'xx/Sming/Sming/third-party/rboot'
C+ app/application.cpp
CC /xx/Sming/Sming//third-party/rboot/appcode/rboot-bigflash.c
CC /xx/Sming/Sming//third-party/rboot/appcode/rboot-api.c
AR out/build/app_app.a
LD out/build/app_0.out
out/build/libmain2.a(time.o): In function `millis':
(.irom0.text+0x94): multiple definition of `millis'
/xx/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/xx/Sming/Sming/third-party/axtls-8266/replacements/time.c:33: first defined here
out/build/libmain2.a(time.o): In function `micros':
(.irom0.text+0xd0): multiple definition of `micros'
/xx/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/xx/Sming/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
out/build/libmain2.a(time.o): In function `configTime':
(.irom0.text+0x138): multiple definition of `configTime'
/xx/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/xx/Sming/Sming/third-party/axtls-8266/replacements/time.c:34: first defined here
out/build/libmain2.a(time.o): In function `time':
(.irom0.text+0x1dc): multiple definition of `time'
/xx/Sming/Sming//compiler/lib//libsmingssl.a(time.o):/xx/Sming/Sming/third-party/axtls-8266/replacements/time.c:30: first defined here
collect2: error: ld returned 1 exit status
/xx/Sming/Sming//Makefile-rboot.mk:435: polecenia dla obiektu 'out/build/app_0.out' nie powiodły się
make: *** [out/build/app_0.out] Błąd 1

xx -> I replace orginal long path

slaff, 2017-02-05:
I think I know what is happening. Do the following:

```
cd $SMING_HOME
rm -rf compiler/lib/libsming*.a
rm -rf compiler/lib/libaxtls.a
rm -rf third-party/axtls-8266 # Most probably that folder was not patched with the latest patch
make ENABLE_SSL=1 
cd ../samples/Basic_Ssl
make ENABLE_SSL=1
```

TomaszKn, 2017-02-05:
Compile it without any errors.
Tomorrow I'm will test how it works.

Thanks

closed 2017-02-05 by slaff