SmingHub/Sming #932: Timer::isStarted return wrong value

iqbaltek, 2017-02-06:
`isStarted()` function returning `true` after `startOnce()` callback fired, only after I manually call `stop()` then `isStarted()` function returns `false`. I consider this behavior as bug, because the Timer callback only fired once but the status not changed.

anakod, 2017-02-07:
Agreed, looks like bug from end user side of view.

slaff, 2017-02-07:
@anakod @iqbaltek Feel free to send a PR with the fixes.

zgoda, 2017-03-22:
From my point this is not a bug and has completely valid and intuitive association to `.start()` method - if a timer has been started (no matter how many times it is expected to fire its callback, one or more, and no matter if its callback has fired or not yet) then `.isStarted()` will return `true` and `false` otherwise. IOW, "started" property is completely other thing than "running" and "timeToFireCallback", and even "callbackFiredAlready". My proposition would be to differentiate these things in `Timer` class at least to 2 properties, `started` and `running`. This way we will have:

- not yet started with `started == false && running == false`,
- started and will fire callback with `started == true && running == true`,
- started and will not fire callback anymore with `started == true && running == false` (it's either stopped or it was set up to fire callback once and callback has been fired already),
- programming error situation with `started == false && running == true`

closed 2020-01-02 by slaff