SmingHub/Sming #937: Not flashing new code

dbelz, 2017-02-10:
Hello Guys,

i have downloaded the Windows Setup over Chocolately and everything seems to be fine. I can run and build the samples and so on. So i have copied one example to another location to write my own code / project.

Before i flashed my own stuff i had the Basic Wifi Sample running on the Board. When i now flash my firmware over eclipse the build process and so on is running fine. But in the Serial Output i see the old messages which are produced by the Basic Wifi example.

How can i fix this? I just want to run my own code.

Kind Regards,
Daniel

ADiea, 2017-02-10:
Please attach console output from the flashing step

dbelz, 2017-02-10:
> 14:27:22 **** Build of configuration Sming for project Mesh ****
make all 
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8374|     884
    rodata|           ReadOnly Data (RAM)|    3FFE8378|    3FFE8DB8|    2624
       bss|      Uninitialized Data (RAM)|    3FFE8DB8|    3FFEF2E8|   25904
      text|            Cached Code (IRAM)|    40100000|    40105C0E|   23566
irom0_text|           Uncached Code (SPI)|    40209000|    402391AA|  197034
Total Used RAM : 29412
Free RAM : 52508
Free IRam : 9220
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
No files found in ./files.
Creating empty out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
Creating empty filesystem.
out/firmware/spiff_rom.bin---------->0x3C000

14:27:31 Build Finished (took 8s.733ms)

>

slaff, 2017-02-10:
@dbelz But, have you run `make flash` to actually re-program your ESP8266 device?

dbelz, 2017-02-10:
yes ofc i have done that. the flash:

`14:45:32 **** Build of configuration Sming for project Mesh ****
make flash 
Killing Terminal to free COM3
FEHLER: Der Prozess "Terminal.exe" wurde nicht gefunden.
/c/Espressif/utils/esptool.exe -p COM3 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x3C000 out/firmware/spiff_rom.bin
Connecting...
Erasing flash...
head: 7 ;total: 7
erase size : 16384

Writing at 0x00000000... (3 %) 
Writing at 0x00000400... (7 %) 
Writing at 0x00000800... (11 %) 
Writing at 0x00000c00... (14 %) 
Writing at 0x00001000... (18 %) 
Writing at 0x00001400... (22 %) 
Writing at 0x00001800... (25 %) 
Writing at 0x00001c00... (29 %) 
Writing at 0x00002000... (33 %) 
Writing at 0x00002400... (37 %) 
Writing at 0x00002800... (40 %) 
Writing at 0x00002c00... (44 %) 
Writing at 0x00003000... (48 %) 
Writing at 0x00003400... (51 %) 
Writing at 0x00003800... (55 %) 
Writing at 0x00003c00... (59 %) 
Writing at 0x00004000... (62 %) 
Writing at 0x00004400... (66 %) 
Writing at 0x00004800... (70 %) 
Writing at 0x00004c00... (74 %) 
Writing at 0x00005000... (77 %) 
Writing at 0x00005400... (81 %) 
Writing at 0x00005800... (85 %) 
Writing at 0x00005c00... (88 %) 
Writing at 0x00006000... (92 %) 
Writing at 0x00006400... (96 %) 
Writing at 0x00006800... (100 %) 
Written 27648 bytes in 2.70 seconds (81.80 kbit/s)...
Erasing flash...
head:  7 ;total: 49
erase size :  172032

Writing at 0x00009000... (0 %) 
Writing at 0x00009400... (1 %) 
Writing at 0x00009800... (1 %) 
Writing at 0x00009c00... (2 %) 
Writing at 0x0000a000... (2 %) 
Writing at 0x0000a400... (3 %) 
Writing at 0x0000a800... (3 %) 
Writing at 0x0000ac00... (4 %) 
Writing at 0x0000b000... (4 %) 
Writing at 0x0000b400... (5 %) 
Writing at 0x0000b800... (5 %) 
Writing at 0x0000bc00... (6 %) 
Writing at 0x0000c000... (6 %) 
Writing at 0x0000c400... (7 %) 
Writing at 0x0000c800... (7 %) 
Writing at 0x0000cc00... (8 %) 
Writing at 0x0000d000... (8 %) 
Writing at 0x0000d400... (9 %) 
Writing at 0x0000d800... (9 %) 
Writing at 0x0000dc00... (10 %) 
Writing at 0x0000e000... (10 %) 
Writing at 0x0000e400... (11 %) 
Writing at 0x0000e800... (11 %) 
Writing at 0x0000ec00... (12 %) 
Writing at 0x0000f000... (12 %) 
Writing at 0x0000f400... (13 %) 
[... 383 lines cut ...]
	for (int i = 0; i < list.count(); i++)
	{
		Serial.print("\tWiFi: ");
		Serial.print(list[i].ssid);
		Serial.print(", ");
		Serial.print(list[i].getAuthorizationMethodName());
		if (list[i].hidden) Serial.print(" (hidden)");
		Serial.println();

		if(list[i].ssid.startsWith("ESP-MESH-")) {
			meshNetworkNodes++;
		}
	}

	Serial.print(meshNetworkNodes);
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE);

	WifiStation.startScan(listNetworks);
}`

But when the Serial Terminal is open it prints a list of the wifis, reconnect try message, scandone message and so on(old things from the Basic Wifi Example)

slaff, 2017-02-10:
> But when the Serial Terminal is open it prints a list of the wifis, reconnect try message, scandone message and so on

Hm, I am getting the feeling that you are seeing the system messages. Can you paste them here?

dbelz, 2017-02-10:
`scandone
List of Wifis here
0scan completed: 17 found
reconnect
f r0, scandone
no PleaseEnterSSID found, reconnect after 1 s
reconnect`

slaff, 2017-02-10:
So my bold guess is that your new code is actually flashed. What you see are system messages. You can try to add the following lines to your program

```
void init()
{
Serial.begin(SERIAL_BAUD_RATE);
Serial.systemDebugOutput(false); // this will disable printing of the system messages

Serial.printf("Great NEW Firmware\n");

// Notice: The next code line below probably will not work if you haven't enabled station mode before that 
//             To fix this uncomment the line below:
//  WifiStation.enable(true);             
WifiStation.startScan(listNetworks);
}`
```

dbelz, 2017-02-10:
It works now! 

Im just starting with this stuff so sorry for the many questions^^

closed 2017-02-10 by slaff