SmingHub/Sming #938: Repeating issues with Windows build

anakod, 2017-02-10:
Unfortunately we have repeating issues with Sming building on Windows OS.

 - https://github.com/SmingHub/Sming/pull/902
 - https://github.com/SmingHub/Sming/pull/929

And currently we have new one:
```
SDK_TOOLS	 ?= $(ESP_HOME)/utils
ESPTOOL		 ?= $(SDK_TOOLS)/esptool.exe
MEMANALYZER  ?= $(SDK_TOOLS)/ESP8266/memanalyzer.exe $(OBJDUMP).exe
```

UDK of previous version use path "C:\Espressif\utils" path for all tools. New version use "C:\Espressif\utils\ESP8266" for it (only Terminal.exe path same as before)

So our makefile willn't build on old version (because MEMANALYZER path wrong), and willn't build on new UDK (because ESPTOOL failed).
For me solution was redefine these variables locally, but it isn't right solution at all.

Аrom a global perspective it's big problem what we have repeating issues with Windows. I think Sming should be able to work on both Linux and Windows after any path.

Few possible ideas/optional solutions to improve situation:

1.  Make new continious integration targert for Windows platform same as MacOS and Linux
2. Create new branch for breaking changes (breaking means what project structure\submodule\building process changed). Adding some examples\bugfixes isn't breaking changes and can be merged to develop directly.
3. Have more tests on breaking changes pull requests. Agreed about minimal time before merging it to develop branch

Let's discuss.

ADiea, 2017-02-10:
I would support latest udk and recommend minimum udk supported version.

anakod, 2017-02-11:
Is Travis CI support windows OS?

slaff, 2017-02-11:
> Make new continious integration targert for Windows platform same as MacOS and Linux

We could use [AppVeyor](https://www.appveyor.com/) for Windows. Can someone of you type here all commands that are needed to install git, MinGW and UDK on Windows? And we can check the compilation of Sming and its tests as a start.

anakod, 2017-02-13:
AppVeyor will be great! But unfortunately my tools installation was done very long time ago. I'm not sure about commands :( I think we should check it on clear virtual machine.

@ADiea 
> Maybe then remove this addition and also the one on memanalyzer and define it by chocolate, and manually for advanced users ?

I don't think what user shouldn't be able to build Sming without additional actions.

Another idea: what reason to support old UDK versions on Windows? Anyway author didn't keep it and rewite installer every time. New UDK package contains also old ESP SDKs so we shouldn't have compatibility issues. I think we can declare support only for UDK 2.0 and above and don't worry about paths problem at all.

ADiea, 2017-02-13:
That's also my initial opinion: why support old UDK. Everyone should update anyway, it contains SDK2.0.0 which will be supported... this will make it easier for new users that start with latest toolchain

anakod, 2017-02-13:
*By the way: p2 and p3 related of coursre not only to windows issues but to any global change affecting building processes and core functionality: https://github.com/SmingHub/Sming/pull/946*

closed 2019-07-04 by slaff