SmingHub/Sming #942: Serial port running at 2400BPS

GarageGadget, 2017-02-11:
Hi,

I need to interface with a device that only supports 2400BPS. It appears that Sming does not support speeds lower than 9600BPS.

Any suggestions appreciated.

slaff, 2017-02-15:
> It appears that Sming does not support speeds lower than 9600BPS.

Can you clarify a bit? It should be possible to set the baud rate to 2400. See the Basic_Serial example and change the start of the init function to look like this
```C
void init()
{
	Serial.begin(2400); // <!--- the value before was SERIAL_BAUD_RATE
```

I haven't tested this on a real device but you can try it. Just make sure to connect to the device using the expected baud rate:

```
python -m serial.tools.miniterm /dev/ttyUSB0 2400
```

GarageGadget, 2017-02-16:
Thanks for the feedback.
 
I my test I was using the Basic_Serial example which I configured to 2400BPS which would run before issuing a software reset. 

I have resolved the issue by patting the watchdog or limiting the amount of data sent to the serial port.

closed 2017-02-16 by GarageGadget