SmingHub/Sming #945: Windows install problem

kwis2, 2017-02-12:
As I have written there was problem with choco install. I have downloaded zip file from  [v 3.1.0](https://github.com/SmingHub/Sming/releases/tag/3.1.0) and with make command I got

```
...
AR out/build/app_app.a
Installing libsming
(!) SSL support is not enabled. To enable it type: 'make clean; make ENABLE_SSL=
1'
Done
Making spiffy utility
CC spiffy.c
In file included from ../third-party/spiffs/src/spiffs.h:14:0,
                 from spiffy.c:4:
../Services/SpifFS/spiffs_config.h:33:36: error: conflicting types for 'uintptr_
t'
    typedef unsigned int            uintptr_t;
                                    ^
In file included from C:/tools/mingw64/x86_64-w64-mingw32/include/stdlib.h:9:0,
                 from spiffy.c:1:
C:/tools/mingw64/x86_64-w64-mingw32/include/crtdefs.h:75:44: note: previous decl
aration of 'uintptr_t' was here
 __MINGW_EXTENSION typedef unsigned __int64 uintptr_t;
                                            ^
make[1]: *** [spiffy.o] Error 1
make: *** [spiffy/spiffy] Error 2
```

anakod, 2017-02-12:
We have related issue before. Probably on different mingw version standard type declarations isn't equal. Can you please copy&paste here part of C:/tools/mingw64/x86_64-w64-mingw32/include/crtdefs.h file with  uintptr_t declaration and lines around that. Or attach that file.

kwis2, 2017-02-12:
here you have crtdefs.h zipped file
[crtdefs.zip](https://github.com/SmingHub/Sming/files/769436/crtdefs.zip)

anakod, 2017-02-13:
You have strange conflict with types declarations. My crtdefs.h didn't contain these declarations.
I've pushed updated Sming version with additional checks. Can you please download it from https://github.com/SmingHub/Sming/tree/fix/uintptr_type_declaration git branch and check? Because for me this test will not be relevant unfortunately.

kwis2, 2017-02-13:
I have checked and now it is new problem with xtensa-lx106-elf-gcc

> C:\tools\sming\Sming>make
> make[1]: Entering directory `/c/tools/sming/Sming/third-party/esp-open-lwip'
> xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FL
> ASH -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals    -c -o lwip/core
> /def.o lwip/core/def.c
> make[1]: xtensa-lx106-elf-gcc: Command not found
> make[1]: *** [lwip/core/def.o] Error 127
> make[1]: Leaving directory `/c/tools/sming/Sming/third-party/esp-open-lwip'
> make: *** [compiler/lib/liblwip_open.a] Error 2
> 
> C:\tools\sming\Sming>

the same problem was with previous versions with `make ENABLE_SSL=1`, but there was no problem with `make`
I think it is path problem. I have instaled sming by Chocolatey, and maybe it would be easier to upgrade environment by modyfy  [choco](https://github.com/SmingHub/Sming/issues/944)?

anakod, 2017-02-13:
I think we already have similar issue before, @slaff what do you think?

PS Probably my fix helps for uintptr_t issue and I will need to make PR with it.

anakod, 2017-02-13:
@kwis2 PLease look topic https://github.com/SmingHub/Sming/pull/933 and folow comments, may be this is reason of your last error?

kwis2, 2017-02-13:
@anakod  I am not exacty understand what do you mean, but sming version I have is the one with merged `--ignore-whitespace` 
I have tried to copy `xtensa-lx106-elf-gcc.exe` into Sming dirs, I got

> C:\tools\sming\Sming>make
> make[1]: Entering directory `/c/tools/sming/Sming/third-party/esp-open-lwip'
> xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FL
> ASH -Os [...]

I don't exactly understand what does it mean, but it was has changed ...

slaff, 2017-02-14:
> PS Also probably my fix helps for uintptr_t issue and I will need to make PR with it.

@anakod Can I suggest something: Please, update your old UDK and MingW to the latest ones, remove the troublesome change (e0753569524a871d1412cfff6d8dd3f26a0f0c89) that you have introduced and create a PR with the fix(es). We have multiple complaints (also on FreeBSD-current) that spiffy is not compiling. The old code was compiling on all platforms, except yours. I guess you would prefer too to have 99% success instead of below 50 % success ;)

anakod, 2017-02-14:
No, I'm using new UDK 2.2.1.  And issue with uintptr_t  resolved, as you can see if you will read above messages attentively. But I will try to update MinGW now, because it can be reason of our difference.

kwis2, 2017-02-15:
I have wondering what could happened and found that problem is with __int64 data type. So I have decided to instal (just copy from another location) 32 bit version of mingw , and sming form  v 3.1.0 zip file compiled without problem.

kwis2, 2017-02-15:
mingw32 solve `make` problem, but `make ENABLE_SSL=1` still dont compile v 3.1.0

> C:\tools\sming\Sming>make clean
> C:\tools\sming\Sming>make ENABLE_SSL=1
> /bin/sh: xtensa-lx106-elf-gcc: command not found
> make[1]: xtensa-lx106-elf-gcc: Command not found
> make[1]: Entering directory `/c/tools/sming/Sming/third-party/axtls-8266'
> xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Wall -Os -g -O2 -Wpointer-arith -Wno-im
> plicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls
> -mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I /c/Espressif/xtensa-lx10
> 6-elf//include -Icrypto -Issl -I.  -c -o crypto/aes.o crypto/aes.c
> make[1]: xtensa-lx106-elf-gcc: Command not found
> make[1]: *** [crypto/aes.o] Error 127
> make[1]: Leaving directory `/c/tools/sming/Sming/third-party/axtls-8266'
> make: *** [compiler/lib/libaxtls.a] Error 2
> 
> C:\tools\sming\Sming>

slaff, 2017-02-15:
Add to your PATH c:\Espressif\/xtensa-lx106-elf\bin\. Try doing that:
```
set PATH=%PATH%;c:\Espressif\/xtensa-lx106-elf\bin\
```

kwis2, 2017-02-15:
`set PATH` command helped.
with current commit a7766da, after I have received:

> C:\tools\sming\Sming>make
> make[1]: Entering directory `/c/tools/sming/Sming/third-party/esp-open-lwip'
> xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FL
> ASH -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals    -c -o lwip/core
> /def.o lwip/core/def.c
> In file included from include/lwip/arch.h:43:0,
>                  from include/lwip/debug.h:35,
>                  from include/lwip/opt.h:46,
>                  from lwip/core/def.c:39:
> include/arch/cc.h:38:21: fatal error: c_types.h: No such file or directory
> compilation terminated.
> make[1]: *** [lwip/core/def.o] Error 1
> make[1]: Leaving directory `/c/tools/sming/Sming/third-party/esp-open-lwip'
> make: *** [compiler/lib/liblwip_open.a] Error 2
> 
> C:\tools\sming\Sming>

slaff, 2017-02-15:
```
cd %SMING_HOME%
make dist-clean
make 
```

kwis2, 2017-02-15:
`make dist-clean` - a lot of warnings because there were no sample dir example below

> make[1]: Entering directory `/C/tools/sming/samples/Ultrasonic_HCSR04'
> /C/tools/sming/Sming/Makefile-project.mk:89: //C/tools/sming/Sming/Makefile-wind
> ows.mk: No such host or network path
> make[1]: stat: //C/tools/sming/Sming/Makefile-windows.mk: No such host or networ
> k path
> make[1]: *** No rule to make target `//C/tools/sming/Sming/Makefile-windows.mk'.
>   Stop.
> make[1]: Leaving directory `/C/tools/sming/samples/Ultrasonic_HCSR04'
> make[1]: Entering directory `/C/tools/sming/samples/Websocket_Client'
> /C/tools/sming/Sming/Makefile-rboot.mk:103: //C/tools/sming/Sming/Makefile-windo
> ws.mk: No such host or network path
> make[1]: stat: //C/tools/sming/Sming/Makefile-windows.mk: No such host or networ
> k path
> make[1]: *** No rule to make target `//C/tools/sming/Sming/Makefile-windows.mk'.
>   Stop.
> make[1]: Leaving directory `/C/tools/sming/samples/Websocket_Client'
> make: *** [samples-clean] Error 2
> 

and `make` wit the same problem

> C:\tools\sming\Sming>make
> make[1]: Entering directory `/c/tools/sming/Sming/third-party/esp-open-lwip'
> xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FL
> ASH -Os [...]

I have checked mingw64 and mingw32 dirs and there is no c_types.h file.

kwis2, 2017-02-15:
I have coppied esp-open-lwip from 3.1.0 zipped versiona and revert to old commit of spiffs (known problem) and sming compiled succes even with `make ENABLE_SSL=1`

closed 2017-02-15 by slaff

anakod, 2017-02-16:
> I have coppied esp-open-lwip from 3.1.0 zipped versiona and revert to old commit of spiffs (known problem) 
Can you provide more details, please?

kwis2, 2017-02-16:
[make problem commit be2b257 #891](https://github.com/SmingHub/Sming/issues/891)