SmingHub/Sming #952: male flash error 2 on linux

dbelz, 2017-02-13:
Hello,

i dropped windows because the choco stuff is using an old version of Sming and i wanted to use the newest available version.

I have now installed everything in an linux environment(ubuntu) but have 1 little problem left:
I can't flash. The build process is running fine and the bins are created. But when i run: sudo make flash i got this:

```
dbelz@esp-dev-machine:/opt/Sming/samples/Basic_WiFi$ make flash
Killing Terminal to free /dev/ttyUSB0
/opt/Sming/Sming/Makefile-project.mk:450: die Regel für Ziel „flash“ scheiterte
make: [flash] Getötet (ignoriert)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin  0x3A000 out/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 32768 @ 0x0... 31744 (96 %)
A fatal error occurred: Timed out waiting for packet header
/opt/Sming/Sming/Makefile-project.mk:450: die Regel für Ziel „flash“ scheiterte
make: *** [flash] Fehler 2
```

Any way to fix this issue? 
I have followed the Linux Quickstart Tutorial. Currently im not using any ide, just the Terminal.

Kind Regards

dbelz, 2017-02-14:
Any suggestions?

avr39-ripe, 2017-02-15:
What is your hardware? Looks like power issue.. or try to update esptool.py from git

slaff, 2017-02-15:
Most probably the USB serial adapter is not working as expected. Set the baud-rate to a lower value. For example `-b 9600`. You might be able to use also a higher value.

dbelz, 2017-02-15:
Thanks the flashing is working with 9600 Baud. But the serial communication not i suggest. I only got "trash" messages and so on^^

Does i need to change the Baud rate also here?

icodk, 2017-02-15:
@dbelz 
IMHO the two things have nothing to do with eachother. Flashing baud is between the programming software(esptool) and esp. When your esp is running the baud should match your serial terminal.

closed 2017-02-20 by slaff