SmingHub/Sming #957: AP network unstabel when both AP and STA are enabled sming 2.1.0 and SDK 1.5.3(aec24ac9)

icodk, 2017-02-14:
If both esp access point(local AP)  and wifi client (to remote AP ex. my router) are enabled, the access point network get's unstable and actually useless. Can't get a full page from the esp web server, Heap is eaten away  and ping timeout most of the time.
If only the AP is enabled it is stable. However it get unstable as soon as the remote access point is configured (SSID to my router) the esp network became useless. Why is it important? Because if you need to reconfigure wifi and the old Ap is not available any longer you must some how (hardware ) reset the old SSID
I am using sming 2.1.0 and SDK 1.5.3(aec24ac9)

slaff, 2017-02-14:
Please, try with the latest Sming version 3.1.0. May be the issue that you are mentioning is fixed already. We have added multiple improvements from 2.1.0 until 3.1.0.

icodk, 2017-02-14:
@slaff Thanks.
Any general hints for upgrading like:
Any reccomendation for sdk (UDK) version ?
My project use espconn_secure_ functions. Should I use lwip ?
Is it needed to use a new version of the compiler  ?
Thanks for help

slaff, 2017-02-15:
> Any reccomendation for sdk (UDK) version ?

Hm, I am Linux user and my preferred SDK is esp-open-sdk, which unfortunately is not available on Windows. 

> My project use espconn_secure_ functions. Should I use lwip ?
If you want to use the custom LWIP with espconn_* functions then you also need to add ENABLE_ESPCONN=1 in your user makefile. For an example take a look at [samples/Basic_SmartConfig/Makefile-user.mk](https://github.com/SmingHub/Sming/blob/develop/samples/Basic_SmartConfig/Makefile-user.mk#L41).

> Is it needed to use a new version of the compiler ?

Not necessarily but if you have one which supports mforce-l32i flag then you will be able to gain some free heap.

closed 2017-02-15 by icodk

avr39-ripe, 2017-02-15:
@icodk Hello! esp8266 has only one RF module, so when you use either STA or AP mode all goes right, but when both STA and AP are enabled problems arise: if esp scan in STA mode it must constantly change channel number so AP functionality becomes unstable, even when esp connects to some AP in STA mode if channel numbers are different it will switch them frequently so you will get instable work. This is feature. It can't be fixed by some Sming/SDK/toolchain update. As a solution you can change your point of view to usage of STA/AP. Try to use just what is really needed, and avoid as much as possible to use both. Look to samples/Basic_WebSkeletonApp as example for STA/AP usage/configuration etc..

icodk, 2017-02-15:
@avr39-ripe Thanks for the insight.
I It all make sense. Esp is a great device but has it's limitations.