SmingHub/Sming #960: Discussion: Replacing esptool2 with esptool.py

slaff, 2017-02-15:
We are using both `esptool2` and `esptool.py`. The latter advanced a lot and has a lot of useful features like:
* faster uploads due to using stubs
* check for ROM overlapping
* etc.

So instead of supporting both of them we can migrate to `esptool.py` only. This will make our Makefiles more consistent and easier to maintain. Is anyone using a feature from `esptool2` that is not present in  `esptool.py`?

slaff, 2017-02-15:
@anakod @ADiea @avr39-ripe Gentlemen, what is your opinion on this?

piperpilot, 2017-02-15:
Need to check with @raburton as I believe there is some magic in esptool2 for the rboot functionality.

avr39-ripe, 2017-02-15:
+1 to @piperpilot I think we have esptool to image generation for rBoot and there is no such functionality in ordinary esptool.py IMHO.. will look on this stuff again, but @raburton can shed the light on it :)

ADiea, 2017-02-15:
I was also thinking about rboot I thibk esp2 has some features for rboot

raburton, 2017-02-15:
esptool2 is required to build rBoot

riban-bw, 2017-02-20:
I too feel we have two tools that overlap significantly but recognise that there are features in each. I did try to write a single tool to cover all Sming requirements but run out of time. Is there an extension to esptool.py that could be added to implement the requirements for rBoot or visa versa?

slaff, 2017-02-20:
>  I did try to write a single tool to cover all Sming requirements but run out of time.

Apart from adding checksums to ROMs, which is hardly used, what are the other `esptoo2` only features that rBoot needs?

> Is there an extension to esptool.py that could be added to implement the requirements for rBoot or visa versa?

Not that I know of but I can help you here.

slaff, 2017-02-21:
@raburton Looking at the comments [here](https://github.com/raburton/rboot/issues/23) it looks as if  we can replace esptool2 with esptool.py. And if no-one is using the checksum feature is there something else that is unique to esptool2 ? Is it in the BIN file generation or in the flashing? 
Thanks in advance.

raburton, 2017-02-21:
Personally I wouldn't want to want to replace a small platform independent program written in C that does the job in one command with a script of half a dozen lines making calls to as many different tools. esptool2 has had a lot of testing and field use, the script posted in that issue might work but really it's a mess and I haven't bothered to test it.

raburton, 2017-02-21:
btw I would strongly recommend everyone use the checksum feature, a better questions is if we should be making that the default

slaff, 2017-02-22:
> a better questions is if we should be making that the default

IMHO, we should make default software one that is advanced and have useful features. Like, for example, faster upload. Or overlapping check. Active community and improvements. I think the discussion is getting too personal for you and I would prefer to decide based on technology, not personal preferences.

closed 2017-02-22 by slaff

raburton, 2017-02-22:
I'm not taking it personally. Your comparison between the two tools isn't really valid, they only share one feature in commons (actual rom creation), which hasn't been discussed. You are comparing 2 different tools with overlapping functionality, not 2 tools that do the same things in slightly different ways.
esptool.py cannot replace esptool2 in building rBoot itself. You could instead replace that necessary feature with a shell script using half a dozen different tools piped together, which may or may not work properly cross platform and isn't very pretty either way.
esptool.py also lacks the checksum feature which is a very good feature of rBoot that users should be using, it adds protection to the flashing / OTA process that simply doesn't exist otherwise, and esptool.py doesn't support this. Yes, we don't use this by default, but if you want to make a decision based on technology this should be a consideration.
The reasons you listed for using esptool.py in the original post are all good features of esptool.py, but they are all advanced features of rom flashing, which esptool2 doesn't do anyway. When using esptool2 you still need something to upload your rom to the device and that is likely to be esptool.py, so you need both, but the fact that esptool.py can do the uploads doesn't compensate for the lack of rBoot specific features that esptool2 provides.

avr39-ripe, 2017-02-22:
@slaff @raburton  Gyus, do not bite ech other! :) I think its all in NAME.. esptool2 was bit bad name, it confuses people that "oh, esptool2 same as esptool.py byt in C and with "2" And this is not true at all.. as you describe in above post it has nothing to compare with esptool.py (ok, almost nothing) IMHO we should include esptool2 as submodule to Sming.. I see no reason not to do so. It is simple (in terms of compilation) tool that simply builds on our target platforms from source, we alreade need host compiller to build spiffy, so why not build esptool2 either and solve meny problems of those who had problems getting esptool by itself??

slaff, 2017-02-22:
Apologies for not being clear. 

It would be great if we have ONE tool for creating ROM images and communicate with the ROM bootloader(for flashing, dumping memory, etc), without sacrificing functionality. That is the goal and I do not have personal preferences. I will be happy also if esptool2 is the winner. If we need two tools, it is  good to know why. Therefore we have this discussion, right?

> Your comparison between the two tools isn't really valid, they only share one feature in commons (actual rom creation), which hasn't been discussed.

So far we have the following stats.

Comparison of ROM creation features
--------------------------------------------------------
|Feature  | esptool.py | esptool | Remarks |
|------------|---------------| -----------| -------------|
| **ROM creation** |
ELF to standalone app | - | + |  Not used right now BUT will come handy in the next versions of Sming
ELF to ROM  image v1.1  | + | + | 
ELF to ROM  image v1.2+ | + | + | esptool.py has '--version=2' option, esptool has -boot2 option.
Adding checksum to ROM image |  - | + | Good optional feature, not widely used at the moment

In that comparison I would say that `esptool2` is the winner due to the support for "ELF to standalone app". Until `esptool.py` has those missing features we can leave `esptool2` in Sming.  And add it as a submodule in the source code + compile it to ease the life of our users. Your thoughts, gentlemen?

avr39-ripe, 2017-02-22:
@slaff as @raburton said in README to esptool2 "The original [tool which inspires esptool2]  has since been updated to v0.0.3 which can write to the flash, but I currently have no intention to add that to esptool2, it is purely a rom creating utility" S I think we must stay with both esptool2 for ROM creation and esptool.py for writing/reading it to and from flash