SmingHub/Sming #969: Memory management stack vs heap

ADiea, 2017-02-18:
Hi everyone.
As we know ( [1](http://tech.scargill.net/esp8266-ram/) [2](http://bbs.espressif.com/viewtopic.php?t=1110) etc ) stack and heap grow from opposite sides and no protection for meeting in the middle exists.
We should find a solution to prevent stack corruption by heap allocating over stack or from going with the stack over previously allocated heap(this is harder/code consuming). 
   * In ummalloc malloc() we should always stop walking when we reach the address of the stack pointer
   * In umm_free_heap_size() -> umm_info() we should also stop walking over the stack and report free heap until the current stack
   * We should make a new function in umm_malloc to calculate available STACK from last occupied heap block until SP and print this info along when a crash happens and we output the stack (which might be corrupted by writing on the heap allocated data) or also this information could be used in certain checkpoints : ex when initiating a SSL request all the SHA/AES etc will allocate on the stack and we can make a check(just an example) 
    * find a way to extract from the compiler the function with the biggest static stack. Try to see if we can disable variable length stack allocation by a compiler flag. Although this will not account for full stack size due to function calling tree we might get a good idea of safe stack size needed and maybe implement something in malloc to stop allocating if we cross a certain stack safeguard.
 
I will start working on a PR for this your input is welcome

ADiea, 2017-02-18:
For printing stack usage [-fstack-usage](https://gcc.gnu.org/onlinedocs/gnat_ugn/Static-Stack-Usage-Analysis.html) can be used. A python should be used to collect all files, sort and output a single file.

slaff, 2017-05-18:
Or we just compile the code using the latest xtensa GCC with stack smashing protection support :) Ask @jcmvbkbc for the details.

jcmvbkbc, 2017-05-18:
> Or we just compile the code using the latest xtensa GCC with stack smashing protection support

It's backported to gcc-5.2.0 and the patch is in the xtensa-1.22.x branch of my crosstool-NG fork. I'll send PR to update esp-open-sdk soon, you can update your ct-ng locally meanwhile. It may be worth building libc/libstdc++ with -fstack-protector*, that would require editing ct-ng configuration, adding -fstack-protector* to the CT_TARGET_CFLAGS.

slaff, 2017-05-19:
> I'll send PR to update esp-open-sdk soon

@jcmvbkbc  That will be great! It will make help us make stack protection default part of the builds.

ADiea, 2017-05-19:
Hopefully this gcc version will also be used in the windows UDK

slaff, 2017-05-21:
Here is a good-old article about stack smashing protection for all of you interested to learn a bit more about it:
http://www.drdobbs.com/security/anatomy-of-a-stack-smashing-attack-and-h/240001832

mikee47, 2018-10-08:
How about raising exception on allocation failures? It would highlight such failures and, if not caught, reset so behaviour would be predictable...