SmingHub/Sming #971: Discussion: Drop support for espressif bootloader?

tius2000, 2017-02-20:
Are there any advantages of the espressif bootloader? Would it make sense to to support rboot only to simplify the framework?

slaff, 2017-02-20:
If no one really insists on supporting Espressif's bootloader we can go even further:
* If bootloaded is needed then only rBoot will be used
* We can offer also images without bootloader, thus freeing space on Flash and memory.

ADiea, 2017-02-20:
+1 to drop support (lathough not used rboot yet, ut if you force me...i'll have to learn :) )

slaff, 2017-02-21:
Moving to only one bootloader, namely the superb rBoot, will help us get rid of:
- [ ] Different LD scripts
- [ ] Different project makefiles
- [ ] Different HttpFirmwareUpdate classes

Adding applications without bootloader, where no OTA or ROM switching is required, will
- [ ] Increase the amount of free space and heap available to the application
 

So I can see a lot of advantages for the code quality and for the users of Sming.

**Is anyone here using the Espressif Bootloader to do Over-The-Air (OTA) firmware updates?**

avr39-ripe, 2017-02-21:
+1 to drop original bootloader
+1 to all @slaff remarks

slaff, 2017-02-21:
> Drop support for espressif bootloader

@anakod @piperpilot @raburton What is your opinion on this?

tius2000, 2017-02-21:
@slaff: great - also the option of having no bootloader at all!

raburton, 2017-02-21:
I don't think I realised we supported the espressif bootloader anyway, or if I did I'd forgotten. In what way is it currently supported?

piperpilot, 2017-02-21:
I use only rboot...I tried the espressif bootloader and it sucked.  I +1 the switch to rboot or nothing.

rosiste, 2017-02-21:
+1 to drop original bootloader

slaff, 2017-02-22:
Ok, starting from version 3.2.0 (#977) we will drop support for Espressif's boolader and use only rBoot as bootloader. Plus, hopefully, add support for standalone applications.

Any final thoughts before we close the discussion?

tius2000, 2017-02-22:
I recently started to create completely refactored makefile for rboot-only projects with many additions (address calculation, automatic linker script creation, dependency detection etc.). However, it is still work in progress. It has a different name, so there are no collisions with the current makefiles. 

Should I create a PR for that when it is finished?

slaff, 2017-02-24:
> Should I create a PR for that when it is finished?

Yes, please.

anakod, 2017-02-25:
We should drop support for old non-actual things, and moreover, we should drop support for my original OTA updater (and than remove Basic_AirUpdate example), because rboot is better and cover all cases.

tius2000, 2017-02-25:
I submitted my refactored makefile as requested #985. However, this is still work in progress, all comments are welcome!

riban-bw, 2017-03-05:
I fully support (+1) making rBoot our only second stage boot loader but please add support for standalone (no second stage boot loader) before dropping support for Espressif second stage boot loader. I am unable to get rBoot and SPIFFS working together (#1009) and we should have a method of creating images without rBoot, just in case of similar issues.

closed 2017-10-12 by slaff