SmingHub/Sming #972: Strange behaviour, random symbols and restarts...

mtzfactory, 2017-02-20:
Hi,

I'm trying to develop a program based in two examples: **HttpServer_ConfigNetwork** and **MqttClient_Hello**, but when I flash it to the esp8266 I get what it seems random symbols in the serial port and it restarts, I think...

https://gist.github.com/mtzfactory/decc1fa356a6ae1da4e109f9e0b78926

You can find the files attached in the link above, you have to copy it into a new copy of HttpServer_ConfigNetwork, in case you want test it, I really want to know if it happen to someone...

I downloaded the new Sming version (3.1.0) and even the dev branch, followed the wiki about **[Application prints random symbols and restarts](https://github.com/SmingHub/Sming/wiki/Application-prints-random-symbols-and-restarts)**, but none has solved my problem...

The super-strange thing is that THE SAME CODE WORKS when I build/flash it with **Sming version 2.1.0**, wow!

**In version 3.1.0**, I can make it work when the line **startMqttClient()** is commented, in the connectOk() function...

I don't think the problem is with the mutt code... I think is something with the size of the rom... is just a guess...

This is the output when it works, with the startMqttClient function commented...
```
#Memory / Section info:
------------------------------------------------------------------------------
out/build/app.out:     file format elf32-xtensa-le
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000384  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00001b28  3ffe8390  3ffe8390  00000470  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00007330  3ffe9eb8  3ffe9eb8  00001f98  2**4
                  ALLOC
  3 .irom0.text   00041b76  4020a000  4020a000  000089f0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00006a50  40100000  40100000  00001f98  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------

# Generating spiffs_bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
out/firmware/spiff_rom.bin---------->0x4C000

18:23:59 **** Build of configuration Sming for project remote-sensor ****
/opt/esptool/esptool.py -p /dev/cu.wchusbserial1a1230 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin  0x4C000 out/firmware/spiff_rom.bin

Flash params set to 0x0200
Compressed 35104 bytes to 24374...

Writing at 0x00000000... (50 %) Writing at 0x00004000... (100 %)
Wrote 35104 bytes (24374 compressed) at 0x00000000 in 2.2 seconds (effective 129.0 kbit/s)...
Hash of data verified.
Compressed 269176 bytes to 192788...

Writing at 0x0000a000... (8 %) Writing at 0x00036000... (100 %)
Wrote 269176 bytes (192788 compressed) at 0x0000a000 in 17.1 seconds (effective 125.7 kbit/s)...
Hash of data verified.
Compressed 196608 bytes to 56473...

Writing at 0x0004c000... (25 %) Writing at 0x00058000... (100 %)
Wrote 196608 bytes (56473 compressed) at 0x0004c000 in 5.0 seconds (effective 312.6 kbit/s)...
Hash of data verified.
```

This is the output when it doesn't work, uncommented startMqttClient function.
```
#Memory / Section info:
------------------------------------------------------------------------------
out/build/app.out:     file format elf32-xtensa-le
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000384  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00001b40  3ffe8390  3ffe8390  00000470  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00007328  3ffe9ed0  3ffe9ed0  00001fb0  2**4
                  ALLOC
  3 .irom0.text   000429ea  4020a000  4020a000  00008a00  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00006a50  40100000  40100000  00001fb0  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------

# Generating spiffs_bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
out/firmware/spiff_rom.bin---------->0x4D000

18:20:34 **** Build of configuration Sming for project remote-sensor ****
Killing Terminal to free /dev/cu.wchusbserial1a1230
/opt/esptool/esptool.py -p /dev/cu.wchusbserial1a1230 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin  0x4D000 out/firmware/spiff_rom.bin

Flash params set to 0x0200
Compressed 35136 bytes to 24388...

Writing at 0x00000000... (50 %) Writing at 0x00004000... (100 %)
Wrote 35136 bytes (24388 compressed) at 0x00000000 in 2.2 seconds (effective 128.9 kbit/s)...
Hash of data verified.
Compressed 272876 bytes to 195276...

Writing at 0x0000a000... (8 %) Writing at 0x00036000... (100 %)
Wrote 272876 bytes (195276 compressed) at 0x0000a000 in 17.3 seconds (effective 125.9 kbit/s)...
Hash of data verified.
Compressed 196608 bytes to 56473...

Writing at 0x0004d000... (25 %) Writing at 0x00059000... (100 %)
Wrote 196608 bytes (56473 compressed) at 0x0004d000 in 5.0 seconds (effective 312.5 kbit/s)...
Hash of data verified.
```

I really need help.

Thanks.

slaff, 2017-02-21:
> I DON'T think the problem is with the mqtt code... I think is something with the size of the rom... is just a guess...

@mtzfactory Please, get the latest version of [esptool.py](https://github.com/espressif/esptool) and try again with it. We added a feature to detect overlapping and also the flash erasing is working as expected.

ADiea, 2017-02-21:
Isn't flash target using esptool2?

slaff, 2017-02-21:
> Isn't flash target using esptool2?

On Windows - yes. On Linux - only for rboot projects, as far as I can tell.

mtzfactory, 2017-02-22:
Hi,

Thanks @slaff for the quick answer, I git cloned the last esptool.py (v2.0-beta2) but the result is still the same... random symbols...

This is the flashing output log...
```
$ make clean && make          
C+ app/application.cpp
C+ app/mqtt_func.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000384  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00001b68  3ffe8390  3ffe8390  00000470  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00006e98  3ffe9ef8  3ffe9ef8  00001fd8  2**4
                  ALLOC
  3 .irom0.text   0004276e  4020a000  4020a000  00008a30  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00006a50  40100000  40100000  00001fd8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/build directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
Adding files in directory 'web/build'.
Added 'bootstrap.css.gz' to spiffs (15615 bytes).
Added 'index.html' to spiffs (12168 bytes).
Added 'jquery.js.gz' to spiffs (30153 bytes).
Added 'settings.html' to spiffs (4639 bytes).
Added 'style.css' to spiffs (7329 bytes).
Added 'wifi-sprites.png' to spiffs (1769 bytes).
out/firmware/spiff_rom.bin---------->0x4D000
 
$ make flashinit && make flash
Flash init data default and blank data.
/opt/esptool/esptool.py -p /dev/cu.wchusbserial1a1230 -b 115200 erase_flash
esptool.py v2.0-beta2
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.3s
Hard resetting...
/opt/esptool/esptool.py -p /dev/cu.wchusbserial1a1230 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x7c000 /opt/esp-open-sdk/sdk/bin/esp_init_data_default.bin 0x4b000 /opt/esp-open-sdk/sdk/bin/blank.bin 0x4D000 /opt/Sming/Sming/compiler/data/blankfs.bin
WARNING: Flash size arguments in megabits like '4m' are deprecated.
Please use the equivalent size '512KB'.
Megabit arguments may be removed in a future release.
esptool.py v2.0-beta2
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
[... 82 lines cut ...]

$ make flashinit && make flash
Flash init data default and blank data.
/opt/esptool/esptool.py -p /dev/cu.wchusbserial1a1230 -b 115200 erase_flash
esptool.py v2.0-beta2
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.2s
Hard resetting...
/opt/esptool/esptool.py -p /dev/cu.wchusbserial1a1230 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x7c000 /opt/esp-open-sdk/sdk/bin/esp_init_data_default.bin 0x4b000 /opt/esp-open-sdk/sdk/bin/blank.bin 0x4B000 /opt/Sming/Sming/compiler/data/blankfs.bin
WARNING: Flash size arguments in megabits like '4m' are deprecated.
Please use the equivalent size '512KB'.
Megabit arguments may be removed in a future release.
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE] [--ucIsHspi]
                           [--ucIsLegacy] [--no-progress] [--verify]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x4b000 for file: /opt/Sming/Sming/compiler/data/blankfs.bin
make: *** [flashinit] Error 2```

slaff, 2017-02-22:
>  it was working before!

Famous last words of developers :) If overlap is detected this means that your first rom is overlapping the second one. Either clean your source code a bit or move the start of the second one to be a bit further from the first one. If you go for the second solution then do the following: 
The linker script and the addresses in the Makefile have to be changed.
You can move the start address from 0xa000 to 0xb000, for example. In the linker irom0_0_seg : org = 0x4020a000, len = (1M - 0xa000) becames irom0_0_seg : org = 0x4020b000, len = (1M - 0xb000)
In the Makefile-project.mk IMAGE_SDK has to be set to 0xb000.
Take a look at that commit for hints: https://github.com/SmingHub/Sming/pull/921/commits/6d9c24ed485ec16c992430547dd17acfb19b18bd

mtzfactory, 2017-02-22:
Je je... yes, famous words... but I don't understand you... I think I missed something...

The first one, with the "startMqttClient" function commented, I can flash the device but it prints random data and it resets itself... 

When I build the same code, this time with the "startMqttClient" function uncommented, when I try to flash it, the **new esptool.py** tells me there is an overlap and it doesn't flash the device.

So, with the first one there is no overlap, even the code size is bigger, but with the second one, smaller code, there is an overlap.

Have you tryed? Does it works for you?
Thanks.

mtzfactory, 2017-02-28:
Hi @slaff,

I followed your instructions, now I can flash the two program versions without problem, but it **still** prints random data and resets itself when I uncomment the **startMqttClient** function...

Please, I need help, have you tried my code?

Thanks in advance...