SmingHub/Sming #974: Websockets can't process large JSON strings (> +/- 1460 bytes)

vjacob, 2017-02-21:
Hello, 

in my application I'm sporadically sending large JSON strings from a Javscript websocket client to the ESP (I'm working with a ESP12F 4Mb). This works fine if the length of the JSON string stays below approximately 1460 bytes. When my JSON string is longer, the WebSocketMessageHandler in my application does report the correct size (message.length() reports the same value as the length of my JSON string that gets sent from the Javascript), however the bytes are scrambled from a certain point in the string. See the example below. The scrambling happens around 1452 bytes, so I guess this is somehow related the LWIP MTU size?  

The is the output I get from printing the string from debug uart1. The total length of the string is 1936, which is the same as the original string generated from the Javasript side. But the last part, after 1452 bytes, contains only scrambled characters. The issue is not debug related, for example because of a tx buffer being too small, because I validate the JSON and it reports an error when I try to parse the string.

`{"des":"lec","src":"app","cmd":"lec.set_show","arg":{"show_idx":0,"show_name":"Test Show_______","schedule_start":"10:30","schedule_end":"11:30","days":127,"looping":1,"fx":[{"idx":0,"name":"Test Item_______","type":"Vegas","speed":100,"duration":1439,"dir":0,"colors":[{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255}]},{"idx":0,"name":"Test Item_______","type":"Vegas","speed":100,"duration":1439,"dir":0,"colors":[{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255}]},{"idx":0,"name":"Test Item_______","type":"Vegas","speed":100,"duration":1439,"dir":0,"colors":[{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255}]},{"idx":0,"name":"Test Item_______","type":"Vegas","speed":100,"duration":1439,"dir":0,"colors":[{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255}]},{"idx":0,"name":"Test Item_______","type":"Vegas","speed":100,"duration":1439,"dir":0,"colors":[{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255}]},{"idx":0,"name":"Test Item_______","type":"Vegas","speed":100,"duration":1439,"dir":0,"colors":[{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255}]},{"idx":0,"name":"Test Item_______","type":"Vegas","speed":100,"duration":1439,"dir":0,"colors":[{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255},{"r":254,"g":255,"b":255}]},{"idx":0,"name":"Test Item_______","type":"Veg°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.N:hƒ°Ñ≈.°Ñ≈.•D¿⁄ë–:%˝j;%•Ñ“⁄áÀ:%…j;%•D¿⁄ë–:%’j;%•D¿⁄È÷:%!j;%•Ñ√⁄áÀ:%-j;%•D¿⁄È“:%9j;%•D¿⁄È‘:%.j;%•D¿⁄âÀ:%°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ö’öëk;%}j;%°Ö’öët;%Ij;%°Ö’öëu;%Uj;%°Ö’öëv;%°k;%°Ö’öëw;%≠k;%°Ö’öëp;%πk;%°Ö’öëq;%°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.°Ñ≈.`

I've started to dig into the Sming websockets code, and I've noticed a comment saying that [splitted frames are not supported](https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/Network/HttpServer.cpp#L129). Is the causing my issue? I would like to investigate further to issue a fix, but I'm not really sure where to start.

Thanks, 
Vincent

avr39-ripe, 2017-02-21:
@vjacob in short: forget about JSON :) Really! just binary data, just hardcore! I mean this, no joking.
Longer answer: if websocket frame is longer then 1460 it reach mtu limit of lwip stack, and with recent update even 1390 is upper limit. Good websocket servers support fragmented frame, but not smings websocket server yet :( But trust me, this isn't your main issue! Even if you CAN receive large JSON strings you will stuck trying parse it inside poor esp ;) JSON encoding/decoding of LARGE JSON strings is VERY memory hungry process! I face such issue with my own project, and my JSON strings were far shorter than yours. As I said in short answer you should switch from json to binary data processing. It is hard, but it is the only reliable solution for large and/or fast data processing on esp8266. When my app on esp and web interface communicate via REST + JSON - it was horribly unreliable, and even switching to websocket+JSON not solve problem completely, only websocket + binary data make app rockstable and MUCH LESS memory hungry :)
But.. support of fragmented ws frames will be good thing to implement in future ;)

vjacob, 2017-02-28:
@avr39-ripe Working with human-readable JSON has proven very transparant for use while working on a project with multiple developers. This way, the correct implementation and bug-fixing of the protocol has been quite painless. I fear this would have taken much more time using a non human-readable format, passed around as byte-arrays. 
For the other comment: there are [JSON parsing libraries ](https://github.com/zserge/jsmn) that take less memory space, and work in more an 'iterator' like fashion, giving start and end positions of tokens in the master byte string.

That being said, I do agree binary data does pose advantages in parsing speed and extensibility. It just wasn't an option in our project. 

I've limited the amount of data to around MTU size for now, but still...any pointers on how to implement multi-frame websocket string support are appreciated.

slaff, 2017-02-28:
Maybe you should change the method below so that is splits the data into multiple frames if length > 1024 bytes 

```
void WebsocketClient::_sendFrame(WSFrameType frameType, uint8_t* msg, uint16_t length)
{
	WebsocketFrameClass wsFrame;
	uint8_t result = wsFrame.encodeFrame(frameType, msg, length, true, true, true);

	if (result && wsFrame._header == nullptr)
	{
		send((char*) &wsFrame._payload[0], wsFrame._payloadLength, false);
	}
	else if (result)
	{
		send((char*) &wsFrame._header[0], wsFrame._headerLength, false);
		send((char*) &wsFrame._payload[0], wsFrame._payloadLength, false);
	}
}
```

vjacob, 2017-02-28:
@slaff Thanks, but that's probably when the ESP is a WebsocketClient connecting to another server? In my project, the ESP is the server and is receiving data from a Javascript client.

rickyitexpert, 2017-07-07:
This problem can be solved by splitting the JSON String in smaller parts with start and end identifiers for JSON on client side. On ESP Side you can reconstruct the splitted strings into one and parse it as JSON.

slaff, 2017-11-28:
> ESP is the server and is receiving data from a Javascript client.

@vjacob Please, check with the latest `develop` version and feel free to reopen this issue if the problem persists.

closed 2017-11-28 by slaff