SmingHub/Sming #975: Spiffs can't mount filesystem after some reboots

itarataiko, 2017-02-21:
Hi,
I'm using Sming ver 3.1.0
On system ready I'm using spiffs_mount_manual for mount file system. After make flash file system is ready and I can read files from file system. Config file loading working well on this stage
Every 60 seconds I try to update some file on the file system and saving current device state into this file.
After some device restarts I see that I cant read config file on the startup phase. I added some code and see that file system is empty and there are no files on it. On after next make flash I can read configuration and save file on the file system.

Can you help me?
Thanks

tius2000, 2017-02-21:
Hi @itarataiko, I noticed similar problems caused by a corrupted flash file system caused by a reboot in the middle of a write operation. The reason might be that you update your files very often.

After I directed the frequent updates to a separate flash sector (without any filesystem) my problems were gone. You should keep in mind that flash memory wears with every erase operation. So it might be a good idea to use different bits for every update (if possible).

itarataiko, 2017-02-22:
Thanks @tius2000 for your help. I'll try to make as you said

closed 2017-02-22 by itarataiko