SmingHub/Sming #977: Preparation for Release 3.2.0

slaff, 2017-02-22:
This issue is created to collect all the tasks we need to do before the 3.2.0

### Source Code
- [ ] Update Sming/SmingCore/SmingCore.h
- [ ] Update Readme.md
- [ ] Merge develop into master
- [ ] Tag the source code in the master branch

### Refactoring
- [ ] Removing all code related to the Espressif bootloader and using rBoot as bootloader
- [ ] Adding options to allow running of ROMs without bootloader (in the cases where the latter is not needed and only takes space )

### Expected PRs
- [ ] Merge Makefile-rboot.mk into Makefile-project.mk 
- [ ] Fixes from ADiea's list (#892)

### Merged PRs
None at the moment.

All merged PRs in this release can be seen from [here](https://github.com/SmingHub/Sming/milestone/10?closed=1)

slaff, 2017-03-10:
@anakod Can you add for the Sming github repo a big yellow or orange label with test "Release Prep" ? This way we could distinguish easily release preps from other issues.

anakod, 2017-03-10:
Yes, I can of course. Can you explain what is it? :)
I'm very like your idea of structuring every release, because it's make process much more transparent, so any improvements here will be helpful, I think.

ADiea, 2017-03-11:
> Simplify FakePgmSpace: if the compiler supports mforce-l32i the functions defined here should match the standard ones. For example strlen_P should be the actual strlen. The compiler will take care to do the l32i magic.
@slaff are you sure you understand mforce-l32i completely? It forces all reads and writes to be aligned. This can be a big performance penality + code size. Also as I understand it only works for multiple of 4 bytes ex if you have 40 bytes I don't quite understand how it will read until byte 40.  So in my opinion this is a tricky compiler option, my windows compiler does not support it and I wouldn't use it anyway as far as I read about it on original post in igrr/esp-arduino

slaff, 2017-03-11:
> Yes, I can of course. Can you explain what is it? :)

@anakod Please, read  [here](https://help.github.com/articles/creating-and-editing-labels-for-issues-and-pull-requests/) how to create "label". Orange color  and text "Release Prep".

slaff, 2017-03-11:
> are you sure you understand mforce-l32i completely? It forces all reads and writes to be aligned. 

@ADiea I guess not :). But I can easily check this. I guess I will have time on Monday to do this. Thanks for bringing that up.

ADiea, 2017-03-13:
Guys just so you know I will be in vacation following next 2 wks, will be unable to fix anything else from the list for the current release :)

closed 2017-04-28 by slaff