SmingHub/Sming #980: flashinit fail with RBOOT_ENABLED = 1

ericogr, 2017-02-22:
hi,

I'm trying to use flashinit with my project but it fail when rboot is enabled.
I'm using v3.1.1 Hotfix and development (11a3912), both generate invalid command to esptool.

testing Sming/samples/Basic_rBoot:

$ make flashinit
Flash init data default and blank data.
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 erase_flash
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Erasing flash (this may take a while)...
Erase took 9.9 seconds
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x3fc000 /opt/esp-open-sdk/sdk/bin/esp_init_data_default.bin 0x3fe000 /opt/esp-open-sdk/sdk/bin/blank.bin  /opt/Sming/Sming/compiler/data/blankfs.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size {detect,4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                           [--no-progress] [--verify]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "/opt/Sming/Sming/compiler/data/blankfs.bin" must be a number
/opt/Sming/Sming/Makefile-rboot.mk:525: recipe for target 'flashinit' failed
make: *** [flashinit] Error 2

slaff, 2017-02-23:
@ericogr Check that PR #981 and report if the solution is working for you.

closed 2017-02-23 by slaff

ericogr, 2017-02-24:
It works now :)
Thanks

PR #981 (2017-02-23): Small fixes to the rBoot makefile.
Fixes #980.
commit 8f14de67 (2017-02-23): Small fixes to the rBoot makefile. (#981)

Fixes #980.