SmingHub/Sming #982: Question regarding free heap

icodk, 2017-02-24:
When connecting to to a remote server using esconnn or espconn_secure, free heap (measured by system_get_free_heap_size()) drops substantially, some times it can take a long time to recover it. Any attempt to connect again  when the free heap size is low can result an exception and a reboot.
Is there a safe way to avoid it it ? 
In a completely different project I used Microchip TCP/IP stack. There, all memory is pre allocated and never released. However it also never case en exception, and I can always start a new connection.
Could anybody shed some light on this issue ?
Thanks

ADiea, 2017-02-24:
At the moment best solution would be to avoid making a new connection when heap is low. While negotiating SSL connection a 16KB is used as decripting buffer. 

In the future we are planning some form of protection to avoid allocating too much heap and causing a reboot, but until then you should never try to alloc last few kb of heap because they are really stack memory and this is causing the reboot (just a guess)

ADiea, 2017-02-24:
In fact we can also make the SSL memory statically allocated this would also improve fragmentation a lot and reuse the buffer for every session. This will however block 16KB of RAM for the life of the app but could be a good option concerning robustness. I think that the buffer is used only in the handshake phase when the master secret is negociated so if you time your connections accordingly you can reuse the SS buffer for many connections. 

@slaff could this work? I see this as a new make option where bm_data is set to a global static variable and the free of the bm_data sets the buffer to unlock state so that a new connection may use it.

slaff, 2017-02-27:
> This will however block 16KB of RAM for the life of the app but could be a good option concerning robustness

Allocating 16K on a device that has roughly 35 K free/usable memory is too much. A buffer of 4K is  more reasonable, although still too high for my taste. You can send a PR to the axtls-8266 project and see if igrr has some better ideas.

closed 2020-01-29 by slaff