SmingHub/Sming #983: Malware? HttpServer_Bootstrap-Example triggers URL:Mal warning in Avast

maerf, 2017-02-24:
HttpServer_Bootstrap-Example downloads serveral files from:

http://simple.anakod.ru/templates/index.html"
http://simple.anakod.ru/templates/bootstrap.css.gz
http://simple.anakod.ru/templates/jquery.js.gz

When I try to download them a URL:Mal is triggered in Avast. 
Its probably a false alert - but then again, warnings about maleware in example code of an IOT-Framework shouldn't be ignored I guess :) 

And its a problem indeed because I can't download the above files without disabling avast.

I couldn't figure out what URL:Mal actually means. Maybe its because it is a dangerous looking russian url? :)

ADiea, 2017-02-24:
anakod is the username of the original creator of this framework. Those files are hosted on his website. It is just an example you can use another website of course :)

maerf, 2017-02-24:
Thank you I have seen that Anakod is a main contributor. But still that doesn't explain the malware warning. I mean, in theory his server could be hacked without him knowing it for example.

anakod, 2017-02-25:
@maerf thank you for notification. It's really false alarm (you can don't trust me of course and check it yourself with other antiviruses).
Can you please send short report to your antivirus web support, because this situation isn't good anyway? It will be helpful for us.

zgoda, 2017-03-01:
Reported: https://support.avast.com/support/tickets/732918, https://support.avast.com/support/tickets/732919, https://support.avast.com/support/tickets/732920

I can't see what's there because I don't have Avast account :D

zgoda, 2017-03-02:
Reply from Avast support:

> Our virus specialists have been working on this problem and it has now been resolved. The provided website isn't detected by Avast anymore.

Please check again.

maerf, 2017-03-02:
Wow that was quick great thanks. My macbook backlight just failed and I'm
back to windows and I can't confirm it now.
But I guess I can sleep soundly tonight then :)

closed 2017-03-02 by maerf