SmingHub/Sming #992: PPP

israellot, 2017-03-02:
LWIP PPP implementation could be included. It would be a nice feature to plug a GPRS modem on serial and talk PPP to it.

slaff, 2017-03-02:
Can you share more details about that?

slaff, 2017-03-02:
I saw some implementations that can be used as a start. So far I can see we need the `sio` functions for the serial communication
```
sio_fd_t 	sio_open (u8_t devnum)
void 	sio_send (u8_t c, sio_fd_t fd)
u8_t 	sio_recv (sio_fd_t fd)
u32_t 	sio_read (sio_fd_t fd, u8_t *data, u32_t len)
u32_t 	sio_tryread (sio_fd_t fd, u8_t *data, u32_t len)
u32_t 	sio_write (sio_fd_t fd, u8_t *data, u32_t len)
void 	sio_read_abort (sio_fd_t fd)
```
which can be mapped easily to our existing HardwareSerial functions. 

And probably we can use as a base something like that https://github.com/MarioViara/gprs. So the more I think about it the more I like the idea. 

The only thing that we need now are volunteers to implement that :) Anyone willing to try it?

israellot, 2017-03-02:
https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/third_party/lwip/netif/ppp
Here's the ppp code on espressif RTOS sdk. It is not present on non-os version. I wonder if that's a starting point.