SmingHub/Sming #993: rBoot makefile does not handle project paths with spaces

riban-bw, 2017-03-02:
I have a project which is stored in a directory which has a space in its path. This builds fine with RBOOT_ENABLED = 0 but fails with RBOOT_ENABLED = 1. This is within MSYS2 (POSIX environment) on Windows 7 64-bit.
It would be advantageous (for debugging this kind of issue) if Makefile-project.mk and Makefile-rboot.mk were better aligned. Currently there are several sections of the same content positioned differently in each makefile.

tius2000, 2017-03-05:
Path and filenames with spaces are a constant source of trouble within make and other tools, I personally try to avoid them whenever possible ...

Perhaps you want to have a look on #971? What is your opinion? Also your feedback on #985 is very welcome ;-)

neut2012, 2017-03-13:
Solution: **Don't use paths with spaces**

Like @tius2000 said, it's not a good practice to use paths with spaces when working with Make files. If not rBoot is the source of trouble, somethings else will nag about the spaces along the way.

riban-bw, 2017-03-13:
I have removed spaces from my project path but this is the 21st century. Operating systems and file systems have handled spaces for decades, as have most of the tools that sit on them. One would hope we don't need to work with ancient file systems forever. I had to deal with all this when I was young. I am hoping for some progress!!!!! Anyway - the point is that the Sming build system can handle paths with spaces so, if we can fix it in the rBoot makefile then we may avoid someone else wasting time chasing their tail like I did!

anakod, 2017-03-13:
I think it will be resolved with new Makefile version.

riban-bw, 2018-01-09:
I just experienced this again (having forgot I had reported the problem in the first place!) because the new version of my IDE has stopped checking for (and removing) spaces in the project path. The last commend from anakod referred to a new Makefile version but that does not seem to have materialised in a form to fix this issue. I think there was some appetite to rationalise the project Makefiles consolidating them to a single one (removing separate Makefile-project and Makefile-rboot) but that hasn't happened and the tow are rather divergent, making it awkward to compare. I will have a look to see if I can identify the cause of this issue because, although I know about GNU make being unfriendly to spaces, it has caught me out a few times and I want Sming to be low friction to all users, including newbies.

slaff, 2019-07-04:
Feel free to re-open this ticket if your issue is still present in the latest `develop` code.

closed 2019-07-04 by slaff