SmingHub/Sming #995: Calling timer rapidly crashes application

riban-bw, 2017-03-03:
If a timer is triggered rapidly, e.g. every 50ms, the application crashes due to watchdog timer after a few timer iterations. Timers should be fail safe and not require user code consideration of limitations. I am guessing (not yet reviewed code) that there is no yield / wdt call within the timer handler routine.

ADiea, 2017-03-03:
I think that if you remain longer than 10ms in a timer callback you must feed the watchdog

slaff, 2017-03-03:
Also if you call a timer every 50ms it can happen that previous and new call overlap. That should also be  taken care of.

riban-bw, 2017-03-03:
Yikes! I would like to decouple the watchdog from timer callback functions to reduce the complexity for developers. I see timers as a core method of providing sequential / loop programming that many developers are familiar with (like the loop function in Arduino) and it should be easy for them. I don't yet know how I trigger events that are longer than 50ms within a timer. I will take a look at the core code over the weekend and see how it works and update docs to match - then consider how we might decouple watchdog.

riban-bw, 2017-03-04:
I have just reworked my code to ensure timer does not crash in to previous call. I have stopped using repeating timer and instead, re-initialise the one-shot timer at the end of the timer call-back routine. This works so my problem was not WDT but instead, as slaff suggested, the callback taking longer than the timer interval. I will add a warning to the API docs and example code to show this alternative method of safe repeating timers. I will close this issue.

closed 2017-03-04 by riban-bw