SmingHub/Sming #1000: build failes with application.o: No such file or directory + chocolatey errors

maerf, 2017-03-04:
I'm been trying to setup a sming on windows for 2,5 days now ever since my mac died, but I can't it to work. I exactly followed the wiki instruactions for chocolatey.
 
**This is what happends when I try to build Blink (or anything else):**

C:\tools\sming.examples\samples\Basic_Blink>make all
C+ app/application.cpp
AR out/build/app_app.a
c:\Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: out/build/app/applica
tion.o: No such file or directory
make: *** [out/build/app_app.a] Error 1

The funny things is: It got over this point **once** and it gave me message saying esptool2 missing (which would have been easy to solve!). But a few minutes later and it failed again. I didn't install anything in between. I think it was the very first build that 'worked' (to esptool2 missing).

I think something went wrong with chocolatey. Attached is the debug ouput of chocolatey. The last attempt was the following command (reinstall according to wiki plus -d):
choco install sming -y -force -d -source 'https://www.myget.org/F/kireevco-chocolatey/'

Mainly its lots of error in ming like the following:
 *** ERROR *** C:\tools\mingw64\/msys/1.0/share/aclocal-1.11/lex.m4: extraction failed
2017-03-04 12:02:52,525 [ERROR] - mingw-get.exe: *** ERROR *** C:\tools\mingw64\/msys/1.0/share/aclocal-1.11/lex.m4: probable package conflict; existing file not overwritten

I googled the "probable package conflict; existing file not overwritten" errors but I couldn't find out how to fix that or if it means. I don't want to break chocolatey even further by manually deleting files.

I'd really appreciate if somebody could help me.

ADiea, 2017-03-04:
Try to install manually maybe

kwis2, 2017-03-06:
As I found choco still link to sming 2.1.0, additionally there is problem with mingw64.
I found that the easiest way, (not exactly optimal) to use sming on win is to install sming by choco, then manually install mingw32 (anywhere), copy files from mingw32 to tools/wmigw64. then delete whole sming dir and download and copy current sming version. Be carefull with case sensitive of folders ie. tools/sming/Sming !!!  then you have to download tools/sming/Sming/third-party from latest develop version of sming - for instance latest version of spiffs won't compile, finally you have to add %PATH to esptool2 and then you are ready to make sming ;)

slaff, 2017-03-06:
@maerf check our [CI script](https://github.com/SmingHub/Sming/blob/develop/appveyor.yml) for hints what we do to get the latest Sming version compiling and running on Windows.

maerf, 2017-03-07:
@kwis2 that solution makes my skin crawl, I hope there is a more long termin solution?

@slaff  I had a look at it but don't see anything obvious. Which doesn't mean there isn't anything.  Any hints about what should I look out for?

I checked back with chocolatey and wrongly filing this following issue. See the attached **log files** there. https://github.com/chocolatey/choco/issues/1183

It turns out the Sming was rejected from chocolatey because of problems in January 2016 already?https://chocolatey.org/packages/sming

I messaged the maintainer kireevco about the problem but I guess you guy already know about this.

After fiddling for days it now compiles. How to savely update to the current version? Do I only have to copy the current sming-master to tools/sming or do I have to touch the c:\expressif as well?

slaff, 2017-04-26:
> Do I only have to copy the current sming-master to tools/sming or do I have to touch the c:\expressif as well?

Only copying the current  sming-master to tools/sming is enough.  c:\expressif  contains the toolchain and for the moment we don't need to have the latest version.

slaff, 2017-04-26:
@maerf If you know someone who can take care for updating and producing new choco packages for Sming we will be happy to help him.

slaff, 2017-11-13:
Feel free to reopen if the issue is still present.

closed 2017-11-13 by slaff