SmingHub/Sming #1003: Order of pinMode and digitalWrite makes a different

icodk, 2017-03-04:
May be a well known fact but for me it was a bit of a pain. 
I use some of the pins as both inputs and outputs, This is possible because it happens in to mutually exclusive states and I don't have enough IOs otherwise. To protect the pin from overcurrent
I wanted to make sure the pin is always low when in output mode so I first called digitalWrite(SS_PIN_1, LOW); and then    pinMode(SS_PIN_1, OUTPUT);
This didn't work but  was hard to find because I have another mutually exclusive function that use the same pin as input (long story). It start working only after I changed the order to
pinMode
digitalWrite
The time is so short  or even none existent so there is not much risk (I think)

Any insight about the issue will be appreciated

ADiea, 2017-03-05:
use hardware protection. Ex a resistor or diodes.
See the attached picture. R25 protects esp device. If esp sets output as low but button is pressed at the sae time then esp will sink no more than 3.3V/1k (value of r25)
You would probably need a transistor to drive the led + pressing the button will also light up the led, but the essential is the resistor close to esp! 
[image]

icodk, 2017-03-05:
@ADiea Thanks for your input. In my case it is not an issue because the button pulls the pin to  GND.
also have protection resistor of 100R, witch is not suficiant in most cases but will take some heat if the time is short.
Could be nice if we could set the output level before the pin is set in an output mode

closed 2017-03-05 by icodk